ARM-DOE/pyart #505: Conda packages for various open source radar packages

jjhelmus, 2016-03-17:
This issue is not precisely Py-ART related but I wanted to keep a check list of all the open source weather radar packages that I want to create conda packages for on [conda-forge](https://conda-forge.github.io/).

## Py-ART and dependencies

* [x] Py-ART
* [x] TRMM RSL
* [ ] CyLP (and a version which supports Python 3?)
* [ ] CBC
* [ ] cvxopt (with GLPK support)
* [ ] GLPK (may need multiple version)
* [ ] pyglpk (and the fork which supports Python 3?)

## Py-ART "extensions"

* [x] ARTView
* [ ] PyTDA
* [ ] SingleDop
* [ ] DualPol
* [ ] PyBlock

## Other open source radar tools
* [x] wradlib
* [ ] MMM-Py
* [ ] CSU_RadarTools
* [ ] RadX

I'm leaving BALTRAD off this list for now, making conda packages of all the components of BALTRAD would be quite a undertaking.

Once complete is should be possible to install any of these packages in conda using `conda install -c conda-forge package_name`.

tjlang, 2016-03-24:
Let me know if I can help with any of the packages I am responsible for.

FYI, I think CSU_RadarTools is not Windows-compatible because it has a Fortran component to speed up the KDP processing, although there is a slower, non-Fortran branch available on its GitHub.

jjhelmus, 2016-03-29:
@tjlang My focus right now is trying to get the GLPK related packages up but I'll give you a shout when I am ready to package up your packages.  If you could tag a release in each of them that would help a lot as it would give a target to point conda build at.

tjlang, 2016-03-30:
@jjhelmus OK, will work on a release tag. Probably won't happen until late April, however.

zssherman, 2024-05-16:
Closing this in favor for an updated version of this list.

closed 2024-05-16 by zssherman