ARM-DOE/pyart #509: Should corrected radial velocity inherit _FillValue?

kirknorth, 2016-03-20:
If a `GateFilter` is used in the `dealias_region_based` or `dealias_unwrap_phase` algorithms, the returned field dictionary will (1) not contain `_FillValue` metadata and (2) the `fill_value` of the data array will be set to `np.ma.default_fill_value`. In my opinion the returned field dictionary should inherit this information from the input velocity field.

jjhelmus, 2016-03-23:
Agreed, the `_FillValue` entry in the field dictionary and the **MaskedArray** `fill_value` attribute should be inherited from the input velocity field.  I'll put a PR together that fixes this for both algorithms.

I should also warn that trying to keep the dictionary entry and the `fill_value` attribute synced can be very difficult and in Py-ART we have not tried very hard to keep the in sync.  From my experience the `fill_value` attribute of **MaskedArrays** are brittle and may change when certain operations are applied to the array.  Where these two properties can have the same value (at least initially), as in this case, the default should be to set them to the same value.

jjhelmus, 2016-03-23:
Also, in instances where the two properties disagree I would consider the `_FillValue` dictionary entry to be the correct value.

closed 2016-03-23 by jjhelmus

kirknorth, 2016-03-24:
@jjhelmus yes I find it somewhat annoying that certain operations (e.g, `np.ma.sqrt`) on a masked array can change its `fill_value` attribute without warning. I would assume this issue has been brought up to the NumPy community, currently there are over 1,000 issues opened on its GitHub repo :stuck_out_tongue_closed_eyes:

scollis, 2016-03-24:
Come to SciPy @knorth ! Give a lightning talk and let your voice be heard

jjhelmus, 2016-04-01:
> I find it somewhat annoying that certain operations (e.g, np.ma.sqrt) on a masked array can change its fill_value attribute without warning

This has actually been fixed in NumPy 1.11.0

``` Python
import numpy as np
print(np.__version__)
a = np.ma.arange(10, fill_value=88)
print(a.fill_value)
print(np.ma.sqrt(a).fill_value)
```

NumPy 1.11.0
```
1.11.0
88
88
```

Old broken behavior in NumPy 1.10.4
```
1.10.4
88
1e+20
```

Now I'm sure there are still more issues with fill_value but at least things are improving!

commit 5d3420b6 (2016-03-23): BUG: dealias_unwrap_phase sets _FillValue

The dealias_unwrap_phase function sets the _FillValue entry of the returned
field dictionary as well as setting the fill_value of the data within the
dictionary from the _FillValue of the input velocity array or the default fill
value when this entry is missing.

closes #509