ARM-DOE/pyart #515: Add additional test to KDP processing module

kirknorth, 2016-04-05:
As discussed in PR #508 it would be a good exercise to test the veracity of `kdp_maesaka` to noisy differential phase profiles in the form of an additional test. @jjhelmus you can assign this issue to me if you would like; I'll submit a PR within the next 5-7 days.

jjhelmus, 2016-04-06:
Assigned.  If you plan to use random data to generate noise be sure to seed the random number generator to get reproducible results between runs.

kirknorth, 2016-04-12:
Since this test turned out to be more difficult to decipher than originally intended (which isn't a bad thing), I'll share my work flow here. I'm basically creating a fake radar with a PSIDP profile that produces constant KDP of 2 deg/km and adding random noise with a defined standard deviation (scale).

```python
# create fake radar (max range limited to 1 km)
radar = sample_objects.make_empty_ppi_radar(101, 1, 1)

# differential phase profile + noise with std = 0.5 deg
np.random.seed(seed=1)
psidp = np.linspace(0.0, 0.004 * 1000.0, radar.ngates)
psidp += np.random.normal(loc=0.0, scale=0.5, size=psidp.size)

# add differential phase field
psidp_dict = get_metadata('differential_phase')
psidp_dict['data'] = np.atleast_2d(psidp)
radar.add_field('differential_phase', psidp_dict)
```

The boundary conditions turned out to be poorly defined using the default settings where `n=20`,

[image]

Switching `n=3` produced better results, but this is likely only applicable to this idealized test case and using a larger sample size is recommended with real data.

[image]

Results using default settings except setting `n=3` and `check_outliers=False`,

[image]

Same as above but increasing the low-pass filter constraint weight to `Clpf=1.0e6` (default is 1.0),

[image]

kirknorth, 2016-04-12:
Note that for `Clpf=1.0` and `Clpf=1.0e6', the mean retrieved KDP is 1.98 deg/km and 2.01 deg/km, respectively.

scollis, 2016-04-12:
aka an insignificant difference.. 

Dr Scott Collis
Purveyor of lofty observations (Atmospheric Scientist) 
Environmental Science Division
Argonne National Laboratory
Mb: +1 630 235 8025
Of: +1 630 252 0550
Become a Py-ART user today! http://arm-doe.github.io/pyart/ <http://arm-doe.github.io/pyart/>

kirknorth, 2016-04-12:
Except when you look at the actual profiles of KDP. The standard deviations are indicative of this: 2.57 deg/km (`Clpf=1.0e0`) versus 0.47 deg/km (`Clpf=1.0e6`).

scollis, 2016-04-12:
Ahh.. right. 

Then you need better tests of goodness :) 

Dr Scott Collis
Purveyor of lofty observations (Atmospheric Scientist) 
Environmental Science Division
Argonne National Laboratory
Mb: +1 630 235 8025
Of: +1 630 252 0550
Become a Py-ART user today! http://arm-doe.github.io/pyart/ <http://arm-doe.github.io/pyart/>

kirknorth, 2016-04-12:
I'm wondering if my noisy PSIDP profile is realistic. This stems from `make_empty_ppi_radar()` bounding maximum range to 1 km. In my fake radar I have 101 range gates meaning range resolution is very high at 10 m. Are phase jumps of as much as +/- 2 deg realistic within 10 m?

jjhelmus, 2016-04-12:
I need to look into what all the parameters are doing in the actual algorithm before I can give an informed opinion.

One easy thing to try repeating the analysis with a reduced amount of noise added to the signal, say standard deviations of 0.3, 0.1, 0.03 and 0.01?  The extreme low end should retrieve the original 4 degrees/km unless something is very wrong.

zssherman, 2024-05-16:
Closing for now inactivity, can reopen if need be.