ARM-DOE/pyart #517: Map / Topography

gamaanderson, 2016-04-09:
Hello 

what is the current approach to adding an user specific map to a plot?
I'm particular interested in adding topography data (assuming the user has that data) similar to the **plot etopo5 topography** in [basemap examples](http://matplotlib.org/basemap/users/examples.html). That kind of map may the helpful in case of big ground clutters or Anomalous propagation.

jjhelmus, 2016-04-11:
The **RadarMapDisplay** and **GridMapDisplay** classes both store the last map plotted in the `basemap` attribute.  It should be possible to use this attribute to add additional plot elements or layers.  For example the `plot_line_xy` method of the **RadarMapDisplay** class uses this attribute to add a line to the plot.  I can work up an example if desired.

gamaanderson, 2016-04-12:
is the basemap attribute public?

scollis, 2016-04-12:
Yep.
Both grid and radar displays make it available at .basemap 

-sent from a mobile device-

jjhelmus, 2016-04-12:
Yes, the basemap attribute is in the public API,  It is documented in the User Guide's section on the [**RadarMapDisplay** class](http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.graph.RadarMapDisplay.html#pyart.graph.RadarMapDisplay)

gamaanderson, 2016-04-12:
Ok, I will see if this works for me and close this issue when i'm sure I have no more questions.

gamaanderson, 2016-06-22:
It worked, I'm closing this issue.

closed 2016-06-22 by gamaanderson

bdolan44, 2017-08-22:
Can I get an example of how you did this @gamaanderson ? I'd like to add topography to some of my plots in the OLYMPEx region. Thanks!

scollis, 2017-08-22:
FYI @gamaanderson will be at #AMSRadar2017

bdolan44, 2017-08-22:
I'm hoping to have some plots using this for #AMSRadar2017. :D

gamaanderson, 2017-08-22:
actually only a small plot at low resolution, I did not explored it much,
it was more like a proof of concept