ARM-DOE/pyart #525: Unable to read in hdf5 file

bhick001, 2016-04-26:
Hello, Im trying to read in some hdf5 data obtained from the Berlin DWD C-band radar.  Pyart doesn't seem to be able to read in the file, yet both h5py and wradlib can. I have tried gamin_hdf5, and posted the error message below at the end after my code. 

If it helps I have shared the data (found below)
https://drive.google.com/file/d/0B_MbpV247W0xQ3BCeWtSTXdJdHc/view?usp=sharing

Thanks for any advice explanation on why this doesn't work.
---------------------------------------------------------------------------

#Read in the data
radarfile = '/Users/brandonh/Documents/Python/radresearch_20140120230000_10392_pro_vol-op-sidpol-01/' + \
    'mvol_20140120230121_10392_pro_vol-op-sidpol-01_0000'
#h5py
f = h5py.File(radarfile,'r')  

#wradlib
radar = wrl.io.read_generic_hdf5(radarfile)

#pyart
radar = pyart.aux_io.gamic_hdf5('/Users/brandonh/Documents/Python/radresearch_20140120230000_10392_pro_vol-op-sidpol-01/mvol_20140120230058_10392_pro_vol-op-sidpol-01_0000')

---------------------------------------------------------------------------

KeyError                                  Traceback (most recent call last)
<ipython-input-69-0e8579322981> in <module>()
      8 radar_wrl= wrl.io.read_generic_hdf5(radarfile)
      9 
---> 10 radar_pyart=pyart.aux_io.read_gamic('/Users/brandonh/Documents/Python/radresearch_20140120230000_10392_pro_vol-op-sidpol-01/mvol_20140120230058_10392_pro_vol-op-sidpol-01_0000')

/Users/brandonh/anaconda/envs/wradlib1/lib/python2.7/site-packages/pyart/aux_io/gamic_hdf5.pyc in read_gamic(filename, field_names, additional_metadata, file_field_names, exclude_fields, valid_range_from_file, units_from_file, **kwargs)
     91 
     92     # Open HDF5 file and get handle
---> 93     gfile = GAMICFile(filename)
     94 
     95     # verify that all scans are present in file

/Users/brandonh/anaconda/envs/wradlib1/lib/python2.7/site-packages/pyart/aux_io/gamicfile.pyc in __init__(self, filename)
     52         """ initialize object. """
     53         self._hfile = h5py.File(filename, 'r')
---> 54         self.nsweeps = self._hfile['what'].attrs['sets']
     55         self._scans = ['scan%i' % (i) for i in range(self.nsweeps)]
     56         self.rays_per_sweep = self.how_attrs('ray_count', 'int32')

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/Users/ilan/minonda/conda-bld/work/h5py/_objects.c:2579)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/Users/ilan/minonda/conda-bld/work/h5py/_objects.c:2538)()

/Users/brandonh/anaconda/envs/wradlib1/lib/python2.7/site-packages/h5py/_hl/attrs.pyc in __getitem__(self, name)
     50         """ Read the value of an attribute.
     51         """
---> 52         attr = h5a.open(self._id, self._e(name))
     53 
     54         if attr.get_space().get_simple_extent_type() == h5s.NULL:

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/Users/ilan/minonda/conda-bld/work/h5py/_objects.c:2579)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/Users/ilan/minonda/conda-bld/work/h5py/_objects.c:2538)()

h5py/h5a.pyx in h5py.h5a.open (/Users/ilan/minonda/conda-bld/work/h5py/h5a.c:2088)()

KeyError: "Can't open attribute (Can't locate attribute: 'sets')"

kmuehlbauer, 2016-04-26:
@bhick001 

You might try the `pyart.aux_io.read_odim_h5` reader. IIRC the DWD-data goes with ODIM_H5 standard.

kmuehlbauer, 2016-04-26:
The `gamic` reader doesn't work with this data, because it builds on a slightly different data layout within the hdf5.

bhick001, 2016-04-26:
@kmuehlbauer 

Well it was as simple as that. Worked great. thanks

closed 2016-04-26 by bhick001