ARM-DOE/pyart #530: grid_from_radars issue with odim data

joshuass, 2016-05-10:
Hi pyART team,

The regridding function grid_from_radars is producing a strange spiral pattern for odimh5 files, see the following [notebook](http://localhost:8888/notebooks/Dropbox/phd/code/rapic_to_pyart/20151126_MtIsa_dust_storm.ipynb)
[odimh5 file](https://www.dropbox.com/s/k3w39pzwbm8hius/201511261318Mnt_Isa.h5?dl=0)

CAPPI image of the scan: 
[image]

The odimh5 data file processes fine in matlab, so I'm not sure what's going on.

Joshua

joshuass, 2016-05-11:
I had a quick attempt at plotting RHIs from the Mt Isa dataset and it looks like the problem may reside in the odimh5 reader (same with #484 and #445). Volume derived RHI scans look very odd, it appears the read has mixed up the azimuth and elevation coordinates of the radar gates!
See the following [notebook](https://github.com/uq-crg/rapic_to_pyart/blob/master/20151126_MtIsa_dust_storm2.ipynb)

jjhelmus, 2016-05-11:
@joshuass Could you repost the link to the original notebook and the sample file.  The first seems to be a link to localhost and send second is 404 for me.

jjhelmus, 2016-05-11:
PR #524 added support for reading ODIM H5 containing RHIs but it was really a stab in the dark.  I have yet to find a ODIM H5 ELEV file that is close enough to the standards to test against.

joshuass, 2016-05-11:
Sorry about that, I've fixed the links. The odimh5 file above contains VOL object.
Great to hear you've added RHI support, I'll test it soon.

jjhelmus, 2016-05-11:
Looking at the individual sweeps in the file it looks like each sweep successive sweep is rotated ~45 degree.  Let me dig into this more it a bit to see if the issue is in the file or in how Py-ART is reading it in.

jjhelmus, 2016-05-11:
Okay, so in ODIM H5 files when a sweep's azimuth angles are not explicitly given in the `startazA` and `stopazA` attributes then as per section 5.1 of the standards doc (version 2.2) the polar data should be ordered starting at 0 degrees and proceed clockwise.  The **index** of the first ray collected in time is given by the `a1gate` attribute in the dataset's where group.  I had misread the meaning of the `a1gate` attribute, using it as the first azimuth angle, which is leading to the rotation issue.  I'll have a fix for this and a few Python 3 related ODIM issues shortly.

jjhelmus, 2016-05-11:
With the fixes in PR #531 the gridding seems to be reasonable.

``` Python 
import pyart
import matplotlib.pyplot as plt
import numpy as np

fn = '201511261318Mnt_Isa.h5'
radar = pyart.aux_io.read_odim_h5(fn, file_field_names=True)
grids = pyart.map.grid_from_radars(
    radar, grid_shape=(31, 301, 301),
    grid_limits=((0, 15000.0),(-150000, 150000), (-150000, 150000)),
    fields=['DBZH'], gridding_algo="map_gates_to_grid")

plt.imshow(grids.fields['DBZH']['data'][0], origin='lower')
plt.savefig('figure1.png')

plt.figure()
display = pyart.graph.GridMapDisplay(grids)
display.plot_grid('DBZH', level=3, vmin=-8, vmax=64, cmap='pyart_NWSRef')
plt.savefig('figure2.png')
```

[image]
[image]

jjhelmus, 2016-05-11:
@joshuass Feel free to re-open this issue if the fix does not work for you.

closed 2016-05-11 by jjhelmus

joshuass, 2016-05-12:
Hi Jonathan,

It's working perfectly now. Thanks for the fix!
https://github.com/uq-crg/rapic_to_pyart/blob/master/20151126_MtIsa_dust_storm_nointerp.ipynb

Cheers,
Joshua

PR #531 (2016-05-11): Various fixed for read ODIM H5 files
* Better Python 3 support for reading strings from ODIM H5 files.
* Bug fix effecting the azimuth angles reported for ODIM H5 files.
* Fallback for range determination in ODIM H5 files.
* ODIM H5 RHI files correctly report the azimuth angles and scan_type.

closes #530