ARM-DOE/pyart #533: pyart.io.write_grid depreciated?

swnesbitt, 2016-05-17:
Was doing some gridding today for the first time in awhile - and tried to use `pyart.io.write_grid` and discovered that the output from both `pyart.map.map_gates_to_grid` and `pyart.map.map_to_grid` is simply a dictionary with the field(s) requested and a key named `ROI`.

The `write_grid` function seems to expect more metadata:
```python
AttributeErrorTraceback (most recent call last)
<ipython-input-8-5301624fdb77> in <module>()
      4 fileout='{}/WKR_{}_{}_{}.nc'.format(outpath,filetype,startt.isoformat(),endt.isoformat())
      5 
----> 6 pyart.io.write_grid(fileout,grid)

/data/keeling/a/snesbitt/anaconda2/envs/py3k/lib/python3.5/site-packages/pyart/io/grid_io.py in write_grid(filename, grid, format, write_proj_coord_sys, proj_coord_sys, arm_time_variables, write_point_x_y_z, write_point_lon_lat_alt)
    203     # create dimensions
    204     dset.createDimension('time', None)
--> 205     dset.createDimension('z', grid.nz)
    206     dset.createDimension('y', grid.ny)
    207     dset.createDimension('x', grid.nx)

AttributeError: 'dict' object has no attribute 'nz'
```

Is this expected?  Was thinking of writing a function that uses `xarray` to create cf-compliant netCDF files, but wanted to check first.

```python
print(pyart._debug_info())
Py-ART version: 1.6.0

---- Dependencies ----
Python version: 3.5.1
NumPy version: 1.11.0
SciPy version: 0.17.0
matplotlib version: 1.5.1
netCDF4 version: 1.2.2

---- Optional dependencies ----
TRMM RSL version: b'v1.49'
CyLP: MISSING
PyGLPK version: MISSING
CVXOPT version: 1.1.8
basemap version: 1.0.8
nose version: 1.3.7
None
```

jjhelmus, 2016-05-17:
Use [`pyart.map.grid_from_radars`](http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.map.grid_from_radars.html) which returns a Grid instance.  This output can be written to a file using `pyart.io.write_grid`.  `pyart.map.map_gates_to_grid` and `pyart.map.map_to_grid` are lower level functions which really should not be directly used.  They are called by `grid_from_radars` to produce an intermediate form of the grid.

swnesbitt, 2016-05-17:
Ahh, duh.  I see if I call `gridding_algo = map_gates_to_grid` I can use your optimized gridding code.  

Thanks!

P.S. Hey @stechma2 - if you add this keyword to the PECAN multi-radar code it will make Angelica much more productive. :)

scollis, 2016-05-17:
#winning

Dr Scott Collis
Purveyor of lofty observations (Atmospheric Scientist) 
Environmental Science Division
Argonne National Laboratory
Mb: +1 630 235 8025
Of: +1 630 252 0550
Become a Py-ART user today! http://arm-doe.github.io/pyart/ <http://arm-doe.github.io/pyart/>

closed 2016-05-17 by swnesbitt