ARM-DOE/pyart #534: BUG: pyart.aux_io.read_gamic()

kmuehlbauer, 2016-05-18:
As reported in [pyart-mailinglist](https://groups.google.com/d/msg/pyart-users/BtsRwN3CB-4/deu0mfLHDAAJ) there is some problem with the `read_gamic`-function. It seems, that the reader is trying to get some keys which are not within the files or have slightly other names (az_speed vs. azimuth_speed).

I tried with the provided file:

```python
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-32-365e531063c5> in <module>()
      3 filename = '2014-10-19--03_20_23_00.mvol'
      4 data, metadata = wrl.io.read_GAMIC_hdf5(filename)
----> 5 radar = pyart.aux_io.read_gamic(filename)

/home/kai/anaconda/envs/wradlib_27/lib/python2.7/site-packages/pyart/aux_io/gamic_hdf5.py in read_gamic(filename, field_names, additional_metadata, file_field_names, exclude_fields, valid_range_from_file, units_from_file, **kwargs)
    241 
    242     # instrument_parameters
--> 243     instrument_parameters = _get_instrument_params(gfile, filemetadata)
    244 
    245     gfile.close()

/home/kai/anaconda/envs/wradlib_27/lib/python2.7/site-packages/pyart/aux_io/gamic_hdf5.py in _get_instrument_params(gfile, filemetadata)
    279     dic = filemetadata('pulse_width')
    280     dic['data'] = gfile.sweep_expand(
--> 281         gfile.how_attrs('pulse_width_us', 'float32') * 1e-6)
    282     instrument_params['pulse_width'] = dic
    283 

/home/kai/anaconda/envs/wradlib_27/lib/python2.7/site-packages/pyart/aux_io/gamicfile.py in how_attrs(self, attr, dtype)
    107         """ Return an array of an attribute for each scan's how group. """
    108         return np.array([self._hfile[s]['how'].attrs[attr]
--> 109                          for s in self._scans], dtype=dtype)
    110 
    111     def how_ext_attrs(self, attr):

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/feedstock_root/build_artefacts/work/h5py-2.6.0/h5py/_objects.c:2687)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/feedstock_root/build_artefacts/work/h5py-2.6.0/h5py/_objects.c:2645)()

/home/kai/anaconda/envs/wradlib_27/lib/python2.7/site-packages/h5py/_hl/attrs.pyc in __getitem__(self, name)
     56         """ Read the value of an attribute.
     57         """
---> 58         attr = h5a.open(self._id, self._e(name))
     59 
     60         if attr.get_space().get_simple_extent_type() == h5s.NULL:

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/feedstock_root/build_artefacts/work/h5py-2.6.0/h5py/_objects.c:2687)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/feedstock_root/build_artefacts/work/h5py-2.6.0/h5py/_objects.c:2645)()

h5py/h5a.pyx in h5py.h5a.open (/feedstock_root/build_artefacts/work/h5py-2.6.0/h5py/h5a.c:2175)()

KeyError: "Can't open attribute (Can't locate attribute: 'pulse_width_us')"
```
and with one of ours:

```python
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-10-2a69d6a89775> in <module>()
      3 #filename = '2014-10-19--03_20_23_00.mvol'
      4 data, metadata = wrl.io.read_GAMIC_hdf5(filename)
----> 5 radar = pyart.aux_io.read_gamic(filename)

/home/kai/anaconda/envs/wradlib_27/lib/python2.7/site-packages/pyart/aux_io/gamic_hdf5.py in read_gamic(filename, field_names, additional_metadata, file_field_names, exclude_fields, valid_range_from_file, units_from_file, **kwargs)
    234     scan_rate = filemetadata('scan_rate')
    235     if scan_type == 'ppi':
--> 236         scan_rate['data'] = gfile.ray_header('az_speed', 'float32')
    237     elif scan_type == 'rhi':
    238         scan_rate['data'] = gfile.ray_header('el_speed', 'float32')

/home/kai/anaconda/envs/wradlib_27/lib/python2.7/site-packages/pyart/aux_io/gamicfile.py in ray_header(self, field, dtype)
    134         data = np.empty((self.total_rays, ), dtype=dtype)
    135         for scan, start, end in zip(self._scans, self.start_ray, self.end_ray):
--> 136             data[start:end+1] = self._hfile[scan]['ray_header'][field]
    137         return data
    138 

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/feedstock_root/build_artefacts/work/h5py-2.6.0/h5py/_objects.c:2687)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (/feedstock_root/build_artefacts/work/h5py-2.6.0/h5py/_objects.c:2645)()

/home/kai/anaconda/envs/wradlib_27/lib/python2.7/site-packages/h5py/_hl/dataset.pyc in __getitem__(self, args)
    413             # This is necessary because in the case of array types, NumPy
    414             # discards the array information at the top level.
--> 415             new_dtype = readtime_dtype(self.id.dtype, names)
    416         mtype = h5t.py_create(new_dtype)
    417 

/home/kai/anaconda/envs/wradlib_27/lib/python2.7/site-packages/h5py/_hl/dataset.pyc in readtime_dtype(basetype, names)
     43     for name in names:  # Check all names are legal
     44         if not name in basetype.names:
---> 45             raise ValueError("Field %s does not appear in this type." % name)
     46 
     47     return numpy.dtype([(name, basetype.fields[name][0]) for name in names])

ValueError: Field az_speed does not appear in this type.
```
@jjhelmus I see two problems here. First, similar names in different gamic-setups like  `pulse_width_us` vs `pulse_width_mks` (as in the provided file) or `az_speed` vs. `azimuth_speed` (as inBonnXPol files). This should be easy to solve. Second, totally different keys and meanings like  in BonnXPol `pulse_width` which is just a enum (0 to 3) which means a specific pulswidth (this set of pulsewidth' we would have to give when calling).

I can try to fix this, a little guidance would be nice.

Any ideas, on how to solve this.

kmuehlbauer, 2016-05-18:
Forgot to add, I used latest `arm_pyart` package from `conda-forge` channel on linux.

kmuehlbauer, 2016-05-19:
I've opened a PR here #535 which covers the problem of the naming of `pulse_width` within different gamic flavours.

@jjhelmus Do you think that it is appropriate to do this also for the `az_speed` vs 'azimuth_speed` or do you have any other ideas? 

For the BonnXPol gamic files the user would have to give the pulse_width, while reading. Could we utilize the config-file?

kmuehlbauer, 2016-05-19:
I found one more difference of the BonnXPol data. The `nyquist_velocity` data is missing from the `scanX/how/extended` group. Or can it be derived from the doppler_moment instead, if this is really needed for the radar object.

There is also some naming issues with the moments. I have `UH` instead of `UZ` and `ZH` instead of `Zh`. Can I add them to the `gamic_field_mapping`?

kmuehlbauer, 2016-05-19:
I've added everything to read the file given via mailing list and also  for the BoXPol data. It will need a little cleanup.

jjhelmus, 2016-05-20:
I believe the immediate issue has been closed by #535.  Can this be closed or should it remain open to track other gamic issues?

kmuehlbauer, 2016-05-20:
I'm settled now. Closing. Thanks!