ARM-DOE/pyart #537: netCDF4/write_grid bug?

stechma2, 2016-05-20:
I am working with a radar meshing script to create composites from multiple radar volumes. This script last worked around May of 2015. With no changes made to the code, the script now fails to work with more than one radar site. The only thing that has changed since then (that I am aware of) is that various packages have been upgraded to their latest versions (i.e., pyart, netcdf4, etc...). The code fails upon trying to write the meshed grid out to netCDF using `write_grid`, with the following errors:

```
Traceback (most recent call last):
  File "/data/keeling/a/stechma2/autoComp/AutoRadarComp.py", line 131, in <module>
    autoRadCmp(sys.argv[1:])
  File "/data/keeling/a/stechma2/autoComp/AutoRadarComp.py", line 80, in autoRadCmp
    pyart.io.write_grid(gridPath, mesh_mapped_ALL_3d)
  File "/data/keeling/a/stechma2/anaconda/lib/python2.7/site-packages/pyart/io/grid_io.py", line 259, in write_grid
    ('nradar', 'nradar_str_length'))
  File "/data/keeling/a/stechma2/anaconda/lib/python2.7/site-packages/pyart/io/cfradial.py", line 774, in _create_ncvar
    ncvar[..., :data.shape[-1]] = data[:]
  File "netCDF4/_netCDF4.pyx", line 3926, in netCDF4._netCDF4.Variable.__setitem__ (netCDF4/_netCDF4.c:42077)
ValueError: total size of new array must be unchanged
```

I have already tried various earlier versions of netcdf4, including 1.1.1 which was likely the version I had installed last summer. 

I have not been able to determine if this is simply an issue with my code, or if there is indeed a bug. My  code is included here for reference:

```python
import matplotlib as mpl
mpl.use('Agg') # Avoids plot output to screen to speed things along
import pyart
import getopt
import sys
import time
from os import listdir
from os.path import isfile, join
from matplotlib import pyplot as plt

slash = '/' # Mac/Linux convention

excluded_fields = ['velocity', 'spectrum_width', 'differential_phase', 'differential_reflectivity', 'cross_correlation_ratio']
keep_nsweeps = 4

def autoRadCmp(argv):
    try:
        opts, args = getopt.getopt(argv,'t:w:x:y:',['gridTime=', 'wrkDir=', 'gridLon=', 'gridLat='])
    except getopt.GetoptError:
        sys.exit('Something went wrong with directory assignment.') 
    for opt, arg in opts:
        if opt in ('-t','--gridTime'):
            gridTime = str(arg) #may not need str() here -- test without it...
        if opt in ('-w','--wrkDir'):
            wrkDir = str(arg) #may not need str() here -- test without it...
        if opt in ('-x','--gridLon'):
            gridLon = float(arg)
        if opt in ('-y','--gridLat'):
            gridLat = float(arg)

    truGridT = gridTime.replace('data' + slash, '')
    print "\nWorking on time ", truGridT
    print "\nUsing grid origin of (", gridLat, ", ", gridLon, ")"

    wrkPath = wrkDir + slash + gridTime

    RadFiles = []
    print "\nThe following radar files will be meshed:"
    for f in listdir(wrkPath):
        print f
        if isfile(join(wrkPath,f)) and ('V06' in f):
            RadFiles.append(f)        
    
    radData = {}
    for x in range(0, len(RadFiles)):
        dataFName = wrkPath + slash + RadFiles[x]
        tmpRadName = 'RADAR_{0}'.format(x)
        radData[tmpRadName] = pyart.io.read(dataFName, exclude_fields=excluded_fields)
#        radData[tmpRadName] = (radData[tmpRadName]).extract_sweeps(range(keep_nsweeps))
    
    print "\nThe following fields are included in each file:"
    for radar, value in radData.iteritems():
        print value.fields.keys()

    

    print "\nNow meshing radar data", time.asctime()
    mesh_mapped_ALL_3d = pyart.map.grid_from_radars(
        (radData.values()),  
        (35, 301, 301),
        ((0., 17000.), 
        (-300.*1000., 300.*1000.), 
        (-300.*1000., 300.*1000.)),
        grid_origin=(gridLat, gridLon),
        grid_origin_alt=0.0,
        fields=['reflectivity'],
        refl_field='reflectivity',
        max_refl=100.,
        copy_field_data=True,
        gridding_algo='map_gates_to_grid')
        
            
    gridPath = wrkDir + slash + 'ReflectivityGrids' + slash + truGridT + '.nc'
    print "\nNow writing the output grid. Path: ", gridPath
    pyart.io.write_grid(gridPath, mesh_mapped_ALL_3d)
     
    
    print "\nNow plotting meshed grid", time.asctime()
    display = pyart.graph.GridMapDisplay(mesh_mapped_ALL_3d)
    fig = plt.figure(dpi=500)
    map_panel_axes = [0.05, 0.05, .4, .80]
    x_cut_panel_axes = [0.55, 0.10, .4, .30]
    y_cut_panel_axes = [0.55, 0.50, .4, .30]
    colorbar_panel_axes = [0.05, 0.90, .4, .03]
    
    # parameters
    level = 2
    vmin = -8
    vmax = 64
    ##lat = 38.25
    ##lon = -99.4
    
    # panel 1, basemap and radar reflectivity
    fig.add_axes(map_panel_axes)
    display.plot_basemap(resolution='i')
    display.plot_grid('reflectivity', level=level, vmin=vmin, vmax=vmax, cmap=pyart.graph.cm.NWSRef)
    ##display.plot_crosshairs(lon=lon, lat=lat)
    
    cbax = fig.add_axes(colorbar_panel_axes)
    display.plot_colorbar(label='Eq. Reflectivity Factor (dBZ)', cax = cbax )
    
    ## panel 2, longitude slice.
    #ax2 = fig.add_axes(x_cut_panel_axes)
    #display.plot_longitude_slice('reflectivity', lon=lon, lat=lat,
    #                             cmap=pyart.graph.cm.NWSRef, vmin=vmin, vmax=vmax)
    #ax2.set_xlabel('Distance Across Domain (km)')
    #
    ## panel 3, latitude slice
    #ax3 = fig.add_axes(y_cut_panel_axes)
    #display.plot_latitude_slice('reflectivity', lon=lon, lat=lat,
    #                            cmap=pyart.graph.cm.NWSRef, vmin=vmin, vmax=vmax)
    #
    ## add a title
    #slc_height = mesh_mapped_ALL_3d.axes['z_disp']['data'][level]
    #dts = num2date(mesh_mapped_ALL_3d.axes['time']['data'], mesh_mapped_ALL_3d.axes['time']['units'])
    #datestr = dts[0].strftime('%H:%M UTC on %d %b %Y')
    #title = 'Sliced at ' + str(slc_height) + ' meters agl at ' + datestr
    #fig.text(0.5, 0.9, title)

    plotPath = wrkDir + slash + 'plots' + slash + truGridT + '.png'
    fig.savefig(plotPath)

    plt.close('all')
    
if __name__ == '__main__':
    autoRadCmp(sys.argv[1:])
```

Some debug info:
```
print(pyart._debug_info())
Py-ART version: 1.6.0

---- Dependencies ----
Python version: 2.7.11
NumPy version: 1.11.0
SciPy version: 0.17.0
matplotlib version: 1.5.1
netCDF4 version: 1.2.2

---- Optional dependencies ----
TRMM RSL version: v1.49
CyLP: MISSING
PyGLPK version: MISSING
CVXOPT version: MISSING
basemap version: 1.0.7
nose version: 1.3.7
None
```

scollis, 2016-05-20:
Dan,
Just checking that the file you tried to write to has been removed . 

-sent from a mobile device-

stechma2, 2016-05-20:
@scollis Thank you for suggesting I check this, but I have for sure been removing the previous netcdf file before each new run.

jjhelmus, 2016-05-22:
@stechma2 Py-ART version 1.6.0 changed a few things in the **Grid** class and it turn the layout of how the **Grid** instanced are written to files changed a bit too.  I'm guessing there is still a bug or two lurking  in that code that is causing the issue you are seeing.  I'm about to leave on vacation so will not be able to track this down until I return but I will be sure to include this on the top of my list once I get back at the end of May.

stechma2, 2016-05-22:
@jjhelmus Thank you in advance for looking into this. I should have tried this before, but I just rolled back to Py-ART version 1.5.0 and the code works as expected now. I will use this version for meshing until you are back from vacation and have had a chance to look into things. Enjoy your vacation!

scollis, 2016-05-31:
@Rumpkie is going to have a quick look at it to see if he can fix it before @jjhelmus get back from PyCon

gavi, 2016-06-02:
I am able to reproduce the same issue when trying to write a grid with multiple radars. Single radar works fine. 

Using pyart 1.7.0

scollis, 2016-06-02:
Thanks Gavi.. This helps knowing that it is only an issue for multi-radar.. likely an issue with Metadata.. @rumpkie check this out.. I hope to have time next week and @jjhelmus will be back on board

Dr Scott Collis
Purveyor of lofty observations (Atmospheric Scientist) 
Environmental Science Division
Argonne National Laboratory
Mb: +1 630 235 8025
Of: +1 630 252 0550
Become a Py-ART user today! http://arm-doe.github.io/pyart/ <http://arm-doe.github.io/pyart/>

gavi, 2016-06-03:
Looks like its failing in 
`def _create_ncvar(dic, dataset, name, dimensions):
`
in `cfradial.py`
when name is `radar_name` and `dic` looks like 

`{'long_name': 'Name of radar used to make the grid', 'data': array(['', ''], 
      dtype='|S1')}`

scollis, 2016-06-03:
Thanks.. I can fix this 

-sent from a mobile device-

scollis, 2016-06-03:
Ok! I am narrowing in on this.. Issue seems to be when radar.metadata['instrument_name']['data'] is a zero length string
So when I do a multi-grid of n radar from nexrad you get a zero length n-tuple of names

QUICK FIX for anyone stuck.. say you have three radars
grid.radar_name = {'data': np.array(['a', 'b', 'b'], dtype='|S1'), 'long_name': 'Name of radar used to make the grid'}
pyart.io.write_grid( 'whatever.nc', grid)

will get you out of jail (works for me)

@jjhelmus : Should we put in an exception for zero length radar names OR change the gridding so that when we have no instrument name it puts in 'null' or suchlike into radar_name

gavi, 2016-06-03:
@scollis Thanks! It good for now. Will wait for permanent fix once @jjhelmus is back.

jjhelmus, 2016-06-07:
I have a one line fix for this which I will submit shortly.  The root of the issue is that trying to write a (2, ) array into a (2, 0) shaped array will not work.  If logic is added to force the second dimension to be at least one, so the shape of the array is (2, 1) then writing works fine and the empty radar names is retained.

jjhelmus, 2016-06-07:
@stechma2 @gavi I believe this issue is fixed by the change made in PR #542.  If you are still having seeing this behavior even with these change feed free to reopen this issue.  

Thanks for bringing this to our attention!

closed 2016-06-07 by jjhelmus

stechma2, 2016-06-08:
Thanks for the fix! Tested it a few hours ago and all seems to be good now. Will reopen if anything seems amiss.

gavi, 2016-06-08:
Thank you. I tested also and seems to be fixed.

PR #542 (2016-06-07): BUG: Writing grids with empty radar names
Allow writing of Py-ART Grid objects using write_grid where the radar_name
attribute of the grid contains empty strings.

closes #537
commit fc773f81 (2016-06-07): BUG: Writing grids with empty radar names

Allow writing of Py-ART Grid objects using write_grid where the radar_name
attribute of the grid contains empty strings.

closes #537