ARM-DOE/pyart #545: pyART support for odim AZIM objects (sector PPIs)

joshuass, 2016-06-10:
Hi,

Thanks for the huge effort on getting odim PPI and RHI scans to work under pyART. There is just one more thing remaining, sector PPIs.

Here is an example [file](https://www.dropbox.com/s/795unyd66qnnr3t/uq-xpol_sppi_20151104_074159.h5?dl=0). I've done lots more work to make our files compliant to odimh5. I don't think there are any issues now. 

Sector PPIs are referred to as AZIM objects the odim format... you can find this attribute in:
/what/object = AZIM and /dataset/what/product = AZIM
Since they vary by azimuth, so /dataset/what/prodpar = PPI
the start and stop azimuth information is also in /dataset/what/

I can probably look at testing this feature in a branch around August after my PhD is submitted. If you get a change before me, please give it a go!

Cheers, J

jjhelmus, 2016-06-13:
Support for AZIM objects was a minor change so I went ahead and added it.  Really basic example

``` Python
import pyart
import matplotlib.pyplot as plt

radar = pyart.aux_io.read_odim_h5('./uq-xpol_sppi_20151104_074159.h5')
display = pyart.graph.RadarDisplay(radar)
display.plot('reflectivity', 0)
plt.show()
```
[image]

scollis, 2016-06-13:
Fantastic work @jjhelmus and thanks for the data and help @joshuass

closed 2016-06-13 by jjhelmus

joshuass, 2016-06-13:
Brilliant, thanks @jjhelmus . It's great to have full support for odim now :tada: I can get to work on writing some pyART tutorials for a third year meteorology course at UQ next semester :book:

PR #546 (2016-06-13): ENH: Support for reading ODIM H5 AZIM files
closes #545
commit f6145e41 (2016-06-13): ENH: Support for reading ODIM H5 AZIM files

closes #545