ARM-DOE/pyart #557: Displaying radar data with Basemap arcgis background

chudlerk, 2016-07-26:
I am attempting to display radar data over shaded relief imagery created by the arcgisimage function within Basemap, as demonstrated [here](http://basemaptutorial.readthedocs.io/en/latest/backgrounds.html)

I tried first creating the Basemap, then created a RadarMapDisplay object and set the basemap to the one I just created, but this just leaves me with a blank white background.

`m = Basemap(llcrnrlon=119.46,llcrnrlat=21.62,urcrnrlon=121.406,urcrnrlat=23.43, projection='merc', resolution = 'i', epsg=4269)
m.drawcoastlines()
m.arcgisimage(service='World_Shaded_Relief', xpixels = 1500, verbose= True)
display = pyart.graph.RadarMapDisplay(radar)
display.basemap = m
display.plot_ppi_map('reflectivity', 0, vmin=0, vmax=64.)
`
I've also tried other things like passing the basemap as an argument when I create the RadarMapDisplay object, and trying to add the background with `display.basemap.arcgisimage(service='World_Shaded_Relief', xpixels = 1500, verbose= True) `
but either the background never shows up, or it disappears when the radar data is plotted.

Is this something I should be able to do?

jjhelmus, 2016-07-28:
The line `m = Basemap(llcrnrlon=119.46,llcrnrlat=21.62,urcrnrlon=121.406,urcrnrlat=23.43, projection='merc', resolution = 'i', epsg=4269)` creates a cylindrical equidistant projection.  Unfortunately the basemap package does not provide a method for performing geographic to map projection coordinate transforms for these projections.  Py-ART needs to perform this transforms to be able to plot radar data on the basemap.  Because the coordinate are not transformed no content is added to the plot.  This issue has been identified as a bug in Py-ART, #556, and a check will be added shortly to inform users when such a projection is created.

It may be possible to use a non-cylindrical and use the `arcgisimage` method  to add a relief image but I am not familiar enough with the function to provide a recommendation.

jjhelmus, 2016-07-28:
Closing as this is a repeat of #556

chudlerk, 2016-07-28:
It doesn't appear to work even after changing the map projection...

jjhelmus, 2016-07-28:
What projections have you tried?  Any projections where the projection attribute is `cyl` will likely not work.  You can check this by adding `print(m.projection)` after the basemap is created.

chudlerk, 2016-07-29:
As it turns out, when using the epsg code required for using arcgisimage(), it is that code that defines the projection, not the "projection" argument passed in when creating the Basemap.

Having said that, even after figuring that out, I am still running into problems.

I create the map with 
`m = Basemap(llcrnrlon=119.46,llcrnrlat=21.62,urcrnrlon=121.406,urcrnrlat=23.43, resolution = 'i', epsg=3825)`

Then, print(m.projection) outputs "tmerc"

After creating the radar display object, I then do
`display.basemap = m`

And print(display.basemap.projection) also outputs 'tmerc'.

I can then do `display.basemap.arcgisimage(<snip>)`, which gives me a shaded relief map, as expected.

However, when I then to `display.plot_ppi_map(field)`, the radar image is shifted away from the shaded relief image, as below:
[image]

And when I now print display.basemap.projection, it shows that the projection is now "lcc"

jjhelmus, 2016-08-01:
@chudlerk Try adding the **Basemap** instance to the `plot_ppi_map` method as the basemap argument: 

``` Python
display.plot_ppi_map(field, basemap=m).
```

If that argument is not provided a new map is created.

chudlerk, 2016-08-02:
That did the trick. Working as expected now, thanks.

chudlerk, 2016-08-02:
Sorry to get off the original topic of the thread, but a quick follow-up question:

I want to cycle through several radar scans, just by plotting one after the other in the same figure, clearing the figure each time. Because creating the shaded relief image takes some time, ideally I would just be able to clear the radar display and replot it at my next time, without getting rid of the basemap background. However, I cannot find a way to just clear the radar display. Am I missing something?

jjhelmus, 2016-08-03:
Good to hear that the suggestion fixed your issue.

You can clear a plot, or technically the axes, of the last sweep plotted by removing the last item from the axes's `collections` attribute.  For example

``` Python
import matplotlib.pyplot as plt
...
# plot the first sweep
display.plot_ppi_map(field, 0, basemap=m)
plt.savefig('sweep_0.png')

# remove the first sweep from the axes
ax = plt.gca()
ax.collections.pop(-1)

# plot the second sweep, do not add another colorbar
display.plot_ppi_map(field, 1, basemap=m, colorbar_flag=False)
plt.savefig('sweep_1.png')
```

This does not seem to work if you use `plt.show()` to pop up a interactive window. I'm not familiar enough with those aspects of matplotlib to know why.

closed 2016-08-03 by jjhelmus