ARM-DOE/pyart #564: Airborne plotting broken

nguy, 2016-08-08:
I just realized that the ```AirborneRadarDisplay.plot_sweep_grid``` was inadvertently broken when a combination of things occurred. 
First, AirborneRadarDisplay was made a subclass of RadarDisplay, thus some methods are superseded. 

Second, LazyLoad _get_x_y_z calculates Cartesian space based upon ```pyart.core.transforms.antenna_vectors_to_cartesian```. For a tail radar such as the NOAA P-3 (for which this class was created), the calculation should actually use ```pyart.core.transforms.antenna_to_cartesian_track_relative``` which uses a coordinate transformation for aircraft using rotation.

How do you think this should be fixed? A conditional statement could be placed somewhere along the path (e.g. in ```pyart.core.radar.get_gate_x_y_z``` that would check the platform and assign the correct lazy load gate calculation.

jjhelmus, 2016-08-08:
@nguy The **AirborneRadarDisplay** is overriding the parent **RadarDisplay** class' `_get_x_y_z` method which performs the necessary logic to prepare and calculate the locations of the gates using ` antenna_to_cartesian` or `antenna_to_cartesian_track_relative`.  From my reading of the file the `get_gate_x_y_z` method of the **Radar** class should never be called.  Are you seeing different behavior?

nguy, 2016-08-08:
So from within  **AirborneRadarDisplay*** ```_get_x_z``` is actually called now, so I guess that this would then rely upon the **RadarDisplay** class method? Maybe it's as simple as just modifying the call to correct method.

But I wasn't sure if you wanted a non-standard method call like this within a subclass?

jjhelmus, 2016-08-08:
`_get_x_z` calls `_get_x_y_z` and discards y.  This should be calling the AirBorneRadarDisplay's `_get_x_y_z` which is what should be overridden and seem to be.  Can you share an example of this not working so I have something more concrete to explore this with?

nguy, 2016-08-08:
Well I just put a print statement in each ```_get_x_y_z``` in both **RadarDisplay** and **AirborneRadarDisplay** to see which is used and the **RadarDispaly** is called. So the override appears to not be working.

nguy, 2016-08-08:
By the way I'm using ARTview for browsing some data, I can share a file if you need, but I think the test file should act the same.

jjhelmus, 2016-08-08:
What test file are you using?

nguy, 2016-08-08:
A Sigmet file from the Calwater experiment. I just looked, apparently there are no test files in Py-ART package. I thought we include one a while back.

jjhelmus, 2016-08-08:
Using a P3 Sigmet file you send a while ago with the currently master I see that the `_get_x_y_z` method of **AirborneRadarDisplay** is called and not the method of **RadarDisplay** in the following script:

``` Python
import pyart
radar = pyart.io.read('./nrm140527202446.RAWFW73')
display = pyart.graph.AirborneRadarDisplay(radar)
display.plot_sweep_grid('reflectivity')
```

nguy, 2016-08-08:
Shoot, scratch that due to my feeble brain. Careful with copy and paste. It IS calling the correct method. I need to look at this further, something isn't quite in the output plot.

nguy, 2016-08-08:
You are right, sorry. I'm going to close this until I find the real issue here. Sorry for the waste of time.

closed 2016-08-08 by nguy

jjhelmus, 2016-08-08:
Don't worry about it.