ARM-DOE/pyart #566: Reading a file file gridded from Radx

rrniii, 2016-08-09:
I have used radx to grid some cfradial files. I would now like to read them in with grid_io.

I had a go at adding a few things into translate between the variable names of the two file types and was successful at a gradx gridded file in but I am still unable to plot the data using display.plot_grid.

I believe the issues is in the setting up of the origin_* variables and projection dict as the code crases when I try display.get_basemap().

Attached  #is the file I am trying to open and the ipython notebook I am using to explore this issue.
[notebookandtestfile.zip](https://github.com/ARM-DOE/pyart/files/409343/notebookandtestfile.zip)

nguy, 2016-08-09:
@rrniii I'm not super familiar with the Grid side, but from the error message and looking at the [reference manual](http://arm-doe.github.io/pyart/dev/user_reference/index.html) I think you need to create a basemap instance and add it to the **GridMapDisplay** after you create it. The following added to your example code creates a basemap, but I still don't get any output on the plot.

```
corners = [dset.variables['lon0'][:].min(), dset.variables['lon0'][:].max(), 
                 dset.variables['lat0'][:].min(), dset.variables['lat0'][:].max()]

lon0 = corners[0] + np.abs(corners[1]-corners[0])/2
lat0 = corners[2] + np.abs(corners[3]-corners[2])/2

fig, ax = plt.subplots(1, 1, figsize=(15, 8))
bm = display.plot_basemap(lat_lines=np.linspace(corners[0], corners[1], 5), 
                          lon_lines=np.linspace(corners[2], corners[3], 5), resolution='l',
                          auto_range=False, min_lon=corners[0], max_lon=corners[1], 
                          min_lat=corners[2], max_lat=corners[3], 
                          projection='tmerc', lat_0=lat0, lon_0=lon0, ax=ax)
p = display.plot_grid('dBZ', level=20, ax=ax)
```

rrniii, 2016-08-10:
Thanks @nguy.  Your suggestion bumped me into the correct direction. I now have a solution that works.  It all came to making my own projParams dictionary from the info in the radx file and getting the correct units.  To plot, I still have to set up a basemap (i.e., it does not plot in one step like a grid made with pyart's internal functions. I think this possibly is due to arrays versus lists versus tuples or and possibly some missing info in the proj Params).

I attach an example file from before, a python notebook where I work through all the issues and an altered version of grid_io that was the additional function defined that reads in this file and allows for expansion to further projections.

[readingradxgriddeddatawithpyart_10Aug2016.zip](https://github.com/ARM-DOE/pyart/files/411553/readingradxgriddeddatawithpyart_10Aug2016.zip)

nguy, 2016-08-10:
Glad it's working @rrniii !

@jjhelmus Is it feasible to provide either a conversion utility for RadX to Py-ART? Or does this exist, the documentation for the radx portion isn't very clear. A related thing I am encountering is to figure out a way to pipe Sigmet files through Radx to apply geolocation corrections for mobile platforms. I can do this in a separate step, and this may be the way to go, but wondered about making it more seamless.

nguy, 2016-08-10:
@rrniii I just looked at the examples you provided. How general is the radx_grid routine you used? You might consider submitting this as a pull request so that it becomes part of the base code.

jjhelmus, 2016-08-11:
@rrniii Really nice work!  It looks like Py-ART should be capable of reading in grid files created with RadX.  I tracked down an issue in Py-ART's coordinate transform code, #568, which I think was causing the issue with plotting these grids without providing explicit basemap parameters.  With that fix and a few modification to the code you provided, plotting RadX grids using Py-ART can now be done with:

``` Python
read_radx_grid('./RAINS_uncal_BNG_5km20160607_222951.nc')
display = pyart.graph.GridMapDisplay(grid)
display.plot_basemap()
display.plot_grid('CCOR', level=10)
plt.show()
``` 

I have modified your code a bit and would be happy to add this functionality to Py-ART if you are interested.  If you would rather submit the change yourself I'd be happy to help you in this process also.

jjhelmus, 2016-08-11:
> Is it feasible to provide either a conversion utility for RadX to Py-ART? Or does this exist, the documentation for the radx portion isn't very clear. A related thing I am encountering is to figure [...]

This is the first for supporting Radx grid files in Py-ART.  Radar files written by Radx should be CfRadial compliant and can be read with `pyart.io.read_cfradial`.  In addition, the [pyart.aux_io.read_radx](http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.aux_io.read_radx.html) function can be used to pipe Sigmet/Dorade/etc files through Radx and read the result without needed to drop to the command line.  The source of that function might help you create a function to apply the geolocations corrections using Radx without using a separate step.

nguy, 2016-08-11:
I'll look at how the input command is structured and the possibility of extending the input if needed. Thanks @jjhelmus

jjhelmus, 2016-10-06:
@rrniii Wanted to check on this issue.  I wrote up a function which can read in NetCDF grid files from RadX based on your code.  If you approve I will add it to Py-ART.

scollis, 2017-03-21:
Where are we at on this? I know Brenda Dolan from CSU was interested in this.. @rrniii Did this work? @jjhelmus Did this go in a Pull request some where?

nguy, 2017-08-28:
A ping on this topic, it was raised again at the AMS Open software short course.

rrniii, 2017-08-28:
Sorry for being lame.

Ill get on this. Glad there is interest.

nguy, 2017-08-28:
Being lame is my gig. 
Thanks for looking at this again.

scollis, 2018-05-22:
@rrniii Is this working for you? If I don't hear back I will close this issue in a few days (clean out time)

rrniii, 2018-05-22:
Hey Scott, I thought I had this working. I am away on a filed campaign and crunched for time but I think I can get it up there when back.

zssherman, 2022-11-03:
Added in #1005 closing

closed 2022-11-03 by zssherman