ARM-DOE/pyart #573: Attenuation correction ZPHI

godofyoga, 2016-08-16:
I am trying  to plot Attenuation correction plot for X-Band radar data using Z-PHI method .In radar data file  having following fields.
corrected_reflectivity_horizontal = DBZH
reflectivity_horizontal' = UH
norm_coherent_power' = NCPH
copol_coeff' = RHOHV
proc_dp_phase_shift' = PHIDP

import pyart
import matplotlib.pyplot as plt
import matplotlib as mpl
import numpy as np

filename = '/home/yogesh/20140504/MDV-20140504-083144-PPIVol-001.nc'
radar = pyart.io.read(filename)

font = {'size': 16}
mpl.rc('font', **font)

Az = radar.azimuth['data']
SRi = radar.sweep_start_ray_index['data']
ERi = radar.sweep_end_ray_index['data']
Rrange = (radar.range['data'])/1000
# remove existing corrections

#radar.fields.pop('specific_attenuation') #File dose not having his field so skip
radar.fields.pop('DBZH') #corrected_reflectivity_horizontal = DBZH

# perform attenuation correction
#corrected_reflectivity_horizontal = DBZH
#refl_field='reflectivity_horizontal' = UH
#ncp_field='norm_coherent_power' = NCPH
#rhv_field='copol_coeff' = RHOHV
#phidp_field='proc_dp_phase_shift' = PHIDP

spec_at, cor_z = pyart.correct.calculate_attenuation(radar, 0, refl_field='UH', ncp_field='NCPH', rhv_field='RHOHV',phidp_field='PHIDP')

radar.add_field('specific_attenuation', spec_at)
radar.add_field('corrected_reflectivity_horizontal', cor_z)

# create the plot
fig1 = plt.figure(figsize=(25, 8))
ax1 = fig1.add_subplot(111)
display = pyart.graph.RadarMapDisplay(radar)

# Ray plot of filtered Reflectivity and unfilrered Reflectivity
d_az = 585
Ray_Az = Az[d_az]
print Ray_Az
fig = plt.figure(figsize = [25,10])
ax3 = fig.add_subplot(111)
plt.plot(Rrange,radar.fields['UH']['data'][d_az, :],'b-',linewidth = 2,label='UH')
plt.plot(Rrange,radar.fields['corrected_reflectivity_horizontal']['data'][d_az, :],'r--',linewidth = 2, label='Corrected')
plt.plot(Rrange,radar.fields['PHIDP']['data'][d_az, :],'g-',linewidth = 2,label='PHIDP')
plt.title('Ray_Az = %f'%(Ray_Az),color = 'r')
plt.legend()

In line plot of corrected reflectivity and  uncorrected reflectivity observe their is no improvement in corrected reflectivity but it is less than uncorrected reflectivity.
How can i get z_offset and phase offset of radar.

How can i get valus of a_coef, beta ? can i use values calculated from T-Matrix simulation using DSD or calculate using observed radar data.

please help me,

Thank you.

hhuangmeso, 2016-08-16:
Coefficients can be calculated from  T-Matrix simulation using DSD of your region. The bias of reflectivity and system phase is a calibration issue. Briefly, the  system phase  can be detected using vertical pointing scans or following the method similar as used in the function [kdp_maesaka](https://github.com/ARM-DOE/pyart/blob/master/pyart/retrieve/kdp_proc.py). Reflectivity can be calibrated using self-consistent method proposed by Vivek et al. 2004 or other absolute calibration methods.

scollis, 2016-08-17:
@godofyoga : This is not an issue with Py-ART. I think you asked a similar question in the email list.. The Z_offset is intended to be calculated using a disdrometer or nearby radar and is used for offset removal. The offset in dBZ is simply added to the reflectivity before calculations are carried out.. Closing this issue

closed 2016-08-17 by scollis