ARM-DOE/pyart #584: Support for NEXRAD Level 3, p135, Enhanced Echo Tops

jacaicedob, 2016-08-29:
I have been trying to load NEXRAD Level 3 data for enhanced echo tops (product 135) with no luck. I get an error saying that product 0 is not supported (for some reason it is finding the wrong product code from the file). I have loaded the data on the NOAA WCT software and it does recognize it as product 135, which should be supported according to the read_nexrad_level3() source code. 

You can access the data I want to plot using the HAS #: HAS010801670

Thanks in advance!

jjhelmus, 2016-08-29:
@jacaicedob I'm not able to reproduce the " product 0 is not supported " error you mentioned with Py-ART 1.6 or the latest GitHub version. Can you describe your setup and Py-ART install as well as the exact steps you used?  The output of `python -c "import pyart; pyart._debug_info()"` would be a a great help.

I did run into an issue when I tried to read in the field data using `pyart.io.read(filename, file_field_names=True)`, specifically a TypeError:

```
TypeError: Cannot cast ufunc subtract output from dtype('int64') to dtype('uint8') with casting rule 'same_kind'
```

I'll be submitting a fix for this shortly.  With that fix, I was able to read and plot one of the files in the HAS order using the following script:

``` Python
import matplotlib.pyplot as plt
import pyart
filename = './KJAX_SDUS72_EETJAX_201508272205'
radar = pyart.io.read(filename, file_field_names=True)
display = pyart.graph.RadarDisplay(radar)
display.plot(135, title='Echo Top Height', colorbar_label='height (km)')
plt.show()
```

I should note that the echo type height field does not have a "nice" name in Py-ART and so the field needed to be indexed by the product number (135).  Additionally, the information in to file about if a given echo has been "topped" is not included in the data extracted by Py-ART, only the height of the echo.

jjhelmus, 2016-08-29:
TypeError bug has been fixed in #585

jacaicedob, 2016-08-29:
This is the error I get: "NotImplementedError: Level3 product with code 1 is not supported"

And this is the output of pyart._debug_info():

Py-ART version: 1.6.0

---- Dependencies ----
Python version: 3.5.1
NumPy version: 1.10.4
SciPy version: 0.17.0
matplotlib version: 1.5.1
netCDF4 version: 1.2.2

---- Optional dependencies ----
TRMM RSL version: MISSING
CyLP: MISSING
PyGLPK version: MISSING
CVXOPT version: MISSING
basemap version: 1.0.7
nose version: 1.3.7

jacaicedob, 2016-08-29:
I use anaconda, how can I apply the fix?

jjhelmus, 2016-08-29:
@jacaicedob What operating system are you running?  What line of code are you using exactly which gives this error?

jacaicedob, 2016-08-29:
@jjhelmus I am using Linux Mint 17.3. Here is the full error message.
```python
radar = pyart.io.read(filename, file_field_names=True)
display = pyart.graph.RadarDisplay(radar)
display.plot(135, title='Echo Top Height', colorbar_label='height (km)')
plt.show()

---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
<ipython-input-12-1c13b05d50e1> in <module>()
----> 1 radar = pyart.io.read(filename, file_field_names=True)
      2 display = pyart.graph.RadarDisplay(radar)
      3 display.plot(135, title='Echo Top Height', colorbar_label='height (km)')
      4 plt.show()

/home/jaime/anaconda3/lib/python3.5/site-packages/pyart/io/auto_read.py in read(filename, use_rsl, **kwargs)
    125         return read_chl(filename, **kwargs)
    126     if filetype == 'NEXRADL3':
--> 127         return read_nexrad_level3(filename, **kwargs)
    128 
    129     # RSL supported formats which are also supported natively in Py-ART

/home/jaime/anaconda3/lib/python3.5/site-packages/pyart/io/nexradl3_read.py in read_nexrad_level3(filename, field_names, additional_metadata, file_field_names, exclude_fields, **kwargs)
     76 
     77     # open the file
---> 78     nfile = NEXRADLevel3File(prepare_for_read(filename))
     79     nradials = nfile.packet_header['nradials']
     80     msg_code = nfile.msg_header['code']

/home/jaime/anaconda3/lib/python3.5/site-packages/pyart/io/nexrad_level3.py in __init__(self, filename)
    133             code = self.msg_header['code']
    134             raise NotImplementedError(
--> 135                 'Level3 product with code %i is not supported' % (code))
    136         bpos += 18
    137 

NotImplementedError: Level3 product with code 1 is not supported```

jjhelmus, 2016-08-29:
And this when you try to read the KJAX_SDUS72_EETJAX_201508272205 file?  Can you verify that the file is not corrupt?  The md5 I have for the file is 92f5b7ef8bd99f0b94d122938cb06103.

jacaicedob, 2016-08-29:
@jjhelmus Yes, it is that same file. I just downloaded it again, opened it up in NOAA WCT, but got the same error with pyart. The md5 I got for the file is 578ede007787b18f14df0acd6f7a40ee.

scollis, 2016-08-29:
Question:
As an echo top product don’t we expect this to be in x/y/z coords? Are we reading in a grid product or a radar object product?

Dr Scott Collis
Purveyor of lofty observations (Atmospheric Scientist) 
Environmental Science Division
Argonne National Laboratory
Mb: +1 630 235 8025
Of: +1 630 252 0550
Become a Py-ART user today! http://arm-doe.github.io/pyart/ <http://arm-doe.github.io/pyart/>

jjhelmus, 2016-08-29:
@jacaicedob Where are you sourcing this file from, the NOAA/NCDC HAS server you mentioned or another source.  The HAS server removed any leading padding and should work (partially) with Py-ART 1.6.0.  Other source may include some leading padding which will require a newer version of Py-ART which includes the fix in #540 or a work around.  

You should be able to tell the difference with the command `python -c "f = open('KJAX_SDUS72_EETJAX_201508272205', 'rb'); print(f.read(30))"`.  A "un-padded" file will begin with SDUS74, a padded file will start with something else.  

In either case you will likely need to update you Py-ART version by building from source once #585 is merged.  Instructions for this process can be found in the [README](https://github.com/ARM-DOE/pyart#installing-from-source).

@scollis NEXRAD produce 135, Enhanced Echo Tops, is a Radial Image type product, it is in antenna coordinates.

jacaicedob, 2016-08-29:
@jjhelmus I do have an unpadded file. I will try it again once #585 is merged and if I have issues, I'll open another thread. Thanks!

jjhelmus, 2016-08-29:
@jacaicedob Sounds good. I just merged #585 so you if you clone the repository and build from source you should get this change.  I'm not certain if this will fix your issue but it is worth a try.

jacaicedob, 2016-08-29:
@jjhelmus I installed the latest version from source and was able to read in and plot the data!

For future reference if anyone else finds this thread, the Echo Tops data is in units of kilofeet so I had to convert to km. See below:

```python
temp = radar.fields[135]['data'] * 0.3048  # Convert from kft to km
radar.add_field_like(135, 'echo_tops', temp)
```

closed 2016-08-29 by jacaicedob

jjhelmus, 2016-08-29:
@jacaicedob Great to hear!  I completely missed that the units of the field were kilofeet (what an odd choice of units) thanks for pointing it out.

dopplershift, 2016-08-30:
8-bit data going up to 70000ft. Lots of decisions based in 90s era computing.