ARM-DOE/pyart #597: Constant Altitude Plan Position Indicator (CAPPI)

meteoswiss-mdr, 2016-09-19:
It would be convenient to create a griddisplay class in line with the radardisplay class to be able to plot CAPPIs in km units.

In addition it may be convenient to be able to plot gridmapdisplay and radarmapdisplay in km units.

jjhelmus, 2016-09-19:
@meteoswiss-mdr I agree completely with the recommendation. Thanks for submitting this.  Adding labels.

scollis, 2017-07-13:
So Just checking I understand @meteoswiss-mdr : x and y axes in km instead of lat/lon?

meteoswiss-mdr, 2017-07-14:
Yes, that is what I meant

vlouf, 2019-07-31:
I recently needed to do a CAPPI, so I have coded up a notebook that do just that:
https://github.com/vlouf/radar_cappi

It wouldn't be a lot of work to add this to Py-ART. Strictly speaking, no interpolation is made, just looking for the nearest neighbour at each grid point for a given altitude.

meteoswiss-mdr, 2019-07-31:
My own code for this can be found here:

https://meteoswiss-mdr.github.io/pyrad/_modules/pyrad/graph/plots_vol.html#plot_cappi

mustafaalicelik, 2021-02-12:
I guess this feature hasn't been added to pyart yet? I need some help on this.

zssherman, 2021-02-12:
@mustafaalicelik I'll see if I can get the pyrad implementation provided by meteoswiss into pyart.

mustafaalicelik, 2021-02-14:
Any improvement?

scollis, 2021-02-14:
Can’t you just make a single level grid?

-sent from a mobile device-

zssherman, 2021-02-18:
@mustafaalicelik I'm hoping to have a try at this sometime tomorrow or early next week.

hamidrizvi, 2021-03-16:
@vlouf 
> I recently needed to do a CAPPI, so I have coded up a notebook that do just that:
> https://github.com/vlouf/radar_cappi
> 
> It wouldn't be a lot of work to add this to Py-ART, so I can do this if you want.
> 
> Strictly speaking, no interpolation is made, just looking for the nearest neighbour at each grid point for a given altitude.

Hello Sir,
I am a master's student working on a quantitative precipitation estimation, for my thesis, I need to compute cappi as well as i've to retrieve rainfall from the same!!,
I just want to know few things from your code;
1. Why did you created azimuth, doesn't your file contain that already?
2. When I change the cappi_altitude in the same code, plot comes the same for every altitude, why? (I was trying to input altitude of 2000, but it only worked for 2 and 1  and those are also alike, there should be difference in cappi at 2 km altitude in comparison with 1km)
3. What is this interpolation method called?
Please have a look at my notebook and help me if possible.! 
https://github.com/hamidrizvi/Analyzing-Precipitation-Data/blob/master/CAPPI-Copy23.ipynb

vlouf, 2021-03-16:
Hi @hamidrizvi ,

Frankly I don't remember well, it was quite some time ago.
1/ It's to convert the azimuth angle (North is 0 degree and going clockwise) to mathematical angle (0 degree is x-axis and going counter-clockwise).
2/ Altitude is in meter I believe, you should use 2000, not 2.
3/ Nearest neighbour.

gabrielcassol, 2022-11-10:
Has this CAPPI product already been added to pyart?

zssherman, 2022-11-10:
@gabrielcassol Hello! At the moment, it has not, still on our list for an addition before the next big release.

syedhamidali, 2022-11-10:
@gabrielcassol I would suggest having a look at this, https://github.com/syedhamidali/CAPPI-NETCDF/blob/main/simple_cappi.ipynb

You can make the data gridded, then plot it at whatever height you want. Here I have plotted Max-CAPPI in this example.

mgrover1, 2024-08-27:
@syedhamidali - is there any interest in contributing this to the main Py-ART repo? If not, we can draft something from our side and ping you for review.

syedhamidali, 2024-08-27:
@mgrover1 Sounds good! I’ll get started on a script for plotting max_cappi with any projection. If that aligns with what you’re thinking, awesome. If not, just let me know what you’d like me to focus on instead.

PR #1633 (2024-08-30): ENH: Max CAPPI

- [x] Closes #597
- [x] Tests added
- [x] Examples Added
- [x] Documentation reflects changes: Not sure

So, I have added the plot_maxcappi function to the gridmapdisplay, here is what it is supposed to do

```python
import pyart

grid = pyart.io.read_grid("GRID_KGRR20240119_163853_V06.nc")
display = pyart.graph.GridMapDisplay(grid)

display.plot_maxcappi('reflectivity', vmin=0, vmax=60, add_slogan=True)
```
[image]
PR #1640 (2024-09-03): ADD: Create CAPPI from Radar

- [x] Closes #597 #1639 #985 #980 #1585
- [x] Tests added
- [ ] Documentation reflects changes   ---  Not sure

#### Description:
This PR introduces a new feature to generate a Constant Altitude Plan Position Indicator (CAPPI) from radar data directly within Py-ART. The new `create_cappi` function allows users to specify the altitude and radar fields for CAPPI generation, simplifying the process and making it more accessible.

The implementation draws on existing work, with valuable insights from the `comp_z.py` module by @dopplerchase and @vlouf's [CAPPI notebook [... cut]
closed 2024-08-30 by mgrover1