ARM-DOE/pyart #600: Reading Argentinian H5 files and saving in CfRadial

mrugna, 2016-10-06:
Hi everyone,

this is not an actual issue of PyART but an enhancement proposal separated in two parts:

I updated the sinarame_h5 reader to match the odim_h5 reader. It's mostly the same but there are some 'unique' features from the h5 files. I will soon submit a PR with this stuff.

The second thing is something @swnesbitt tried to pull in #480. Writing cfradial from the h5 files is something I found really useful and I know some people who can use this as well.

This particular issue I found was on the way the original bufr files are created:  the variables are in separated files, they have different gate bins, strategies applied, etc. It's a huge mess to work with this data as is. I know this is a _very_ specific case, but I came up with a script (adapting something Steve has done) that saves all the variables and their metadata in one cfradial file. Should this be added as another function in the `sinarame_h5` file? Or, as discussed in the PR, it will fit better as an example or in the wiki?

I'm polishing that script (because it's in spanish right now) and I'll upload it a gist so you have a better idea of what I'm talking about. In the meantime let me know what you think of this addition.

jjhelmus, 2016-10-12:
@mrugna Looking forward to seeing the updates to the sinarame_h5 reader.

As for a script to help in the reading of h5 files, I don't have a good place to put this but I think the wiki is the best place.  If someone has suggestions for a better place for scripts like this might belong in Py-ART, I'd be glad to hear them.

mrugna, 2016-10-12:
Sorry for the delay. Here is the gist for the cfradial writer.

I know the way I handle the errors is awful but I'm still learning those. I would append it to the sinarame_h5 reader because it's tightly related to it, but now you get a better idea of what I'm talking about.

https://gist.github.com/mrugna/c2d0fa8bb9cece86048bcd5ed3499a87

jjhelmus, 2016-10-14:
@mrugna Looking the function in the gist I think adding that to the sinarame_h5 module in Py-ART itself would be fine.  Feel free to submit a PR adding it.

mrugna, 2016-10-14:
Great! I'll submit the PR today and try not to mess the documentation.

scollis, 2016-10-14:
Thanks Martin!

-sent from a mobile device-

PR #608 (2016-10-17): Added a CfRadial writer for SINARAME H5 files
I believe I didn't forget anything for the dev documentation. Let me know if something's wrong with the addition.

This fully closes #600.
closed 2016-10-17 by jjhelmus