ARM-DOE/pyart #606: Altitude data available from ARM sonde data stream

kirknorth, 2016-10-14:
I'm working on some VAD code and came across this potential issue. Altitude information is available in ARM sonde data streams, yet heights are computed from pressure measurements in [line 59](https://github.com/ARM-DOE/pyart/blob/master/pyart/io/arm_sonde.py#L59) of `read_arm_sonde`. Just curious why this is.  Here's a plot of the differences between the two for a test file [here](http://www.meteo.mcgill.ca/~kirk/). 

[image]

According to the variable attributes, sonde altitude measurements should be in meters above mean sea level, which matches what's written in the docs of `HorizontalWindProfile` class.

jjhelmus, 2016-10-14:
I think the reason why altitude is not being read in is that I did not see that field in the file when I wrote the function.  I was looking for a `height` variable and skipped over the `alt` variable.  I'll submit a PR fixing this shortly.

kirknorth, 2016-10-14:
Okay, just making sure it wasn't an altitude correction fix, calibration issue with original data, or anything of that sort. Thanks @jjhelmus.

PR #607 (2016-10-14): BUG: Read in altitude form ARM sonde files
Read in the altitude directly from ARM sonde files from the alt variable.
Previously the altitude/height was calculated from the pressure which would
give slightly incorrect values.

closes #606
commit 757f143b (2016-10-14): BUG: Read in altitude form ARM sonde files

Read in the altitude directly from ARM sonde files from the alt variable.
Previously the altitude/height was calculated from the pressure which would
give slightly incorrect values.

closes #606
closed 2016-10-14 by jjhelmus