ARM-DOE/pyart #609: reading rainbow version 5.43.10 data

srbrodzik, 2016-10-14:
I've been able to read some version 5.34.16 rainbow files but am unable to do so for a more current rainbow version (5.43.10).  Was wondering if support for these files will be included at some future date.  I've attached an example of the newer version.  

[2015111700500600dBZ.zip](https://github.com/ARM-DOE/pyart/files/530587/2015111700500600dBZ.zip)

Here's the error I get:

In [25]: file = '/home/disk/bob/olympex/xband/rainbowFiles/RAW_DOPVOL1_A/DBZ/2015111700500600dBZ.azi'

In [26]: radar = pyart.aux_io.read_rainbow_wrl(file)                                                 
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-26-6b5fa6247414> in <module>()
----> 1 radar = pyart.aux_io.read_rainbow_wrl(file)

/home/disk/shear2/brodzik/.local/lib/python2.7/site-packages/pyart/aux_io/rainbow_wrl.pyc in read_rainbow_wrl(filename, field_names, additional_metadata, file_field_names, exclude_fields, **kwargs)
    297         moving_angle[ssri[i]: seri[i]+1], angle_start, angle_stop = (
    298             _get_angle(slice_info['slicedata']['rayinfo'],
--> 299                        angle_step=angle_step, scan_type=scan_type))
    300 
    301         # time

/home/disk/shear2/brodzik/.local/lib/python2.7/site-packages/pyart/aux_io/rainbow_wrl.pyc in _get_angle(ray_info, angle_step, scan_type)
    385         if angle_step is None:
    386             raise ValueError('Unknown angle step')
--> 387         angle_start = np.array(ray_info['data']*bin_to_deg, dtype='float64')
    388         if scan_type == 'rhi':
    389             ind = (angle_start > 225.).nonzero()

TypeError: list indices must be integers, not str

jjhelmus, 2016-10-17:
I think there is a missing indexing in the routine that reads in Rainbow files which supply only a starting angle not a starting and ending angle.  Submitting a fix in a moment.

srbrodzik, 2016-10-17:
Thanks Jonathan.  The OLYMPEX folks at NASA and UW will be very happy when
this is fixed.

Stacy

PR #613 (2016-10-20): FIX: fix _get_angle-function
 to read single `startangle` and dual `startangle` and `stopangle`  in multi-element ray_info

Following the discussion in #610 this PR fixes the issue for now.

fixes #609 
supersedes #610
closed 2016-10-20 by jjhelmus