ARM-DOE/pyart #621: Error in NEXRADLevel2File __init__()

normbw, 2016-11-18:
In pyart/io/nexrad_level2.py of the current master, line 147:

``elif compression_or_ctm_info in {b'\x00\x00', b'\t\x80'}:``

raises a syntax error on importing pyart, pointing at the sort-of dictionary initialization.  It looks as if this was introduced in v1.7.0, which makes me wonder if I'm missing something obvious...  Was a dictionary intended?

dopplershift, 2016-11-18:
That's a set literal...which should be fine in Python >= 2.7. What version of Python are you using?

normbw, 2016-11-19:
Ah, that's it.  Python 2.6.6, which the INSTALL.rst indicates is acceptable (2.6.x and newer).  Is mathematical set behavior needed here?  If not, switching to a tuple (obviously) would avoid breaking the compatibility with 2.6.x.

dopplershift, 2016-11-19:
I think set is used here because in situations with many items `in` can be quicker on a set. In this case, I see no reason not to use a tuple. Can you make the change and submit a Pull Request?

closed 2016-11-21 by jjhelmus

jjhelmus, 2016-11-21:
@normbw Thanks for the fix!

As a heads up the current version of Py-ART does not "officially" support Python 2.6. In fact, Py-ART [version 1.6.0](https://github.com/ARM-DOE/pyart/releases/tag/v1.6.0) was the last release that supported Python 2.7.  Support for Python 2.6 and 3.3 was dropper in [version 1.7.0](https://github.com/ARM-DOE/pyart/releases/tag/v1.7.0).  

That said Py-ART may still work with Python 2.6 but the project is no longer testing against that version so additional bugs like this may pop up.  I'm happy to merge in bug fixes if anyone cares to submit them but do not have the time to actively support or test against Python 2.6.  I would also encourage anyone still running Python 2.6 to update to either Python 2.7 or even better Python 3.5.

I'll update the `INSTALL.rst` file shortly to correct the support listing in that file.

jjhelmus, 2016-11-21:
Updated text in `INSTALL.rst` in #623 

Also, thanks @dopplershift for fielding this issue.  I was traveling last week and was very glad to see that this got addressed in my absence.

dopplershift, 2016-11-21:
No worries--it was a straightforward enough question. 😄 

Also, above do you mean 1.6.0 "was the last release that supported Python" **2.6**?

jjhelmus, 2016-11-21:
> Also, above do you mean 1.6.0 "was the last release that supported Python" 2.6?

Yup, will edit the comment to fix this.

PR #622 (2016-11-21): Fix syntax error under python 2.6.x
Convert set literal to tuple.

Resolves: #621
commit fd7f8724 (2016-11-21): Fix syntax error under python 2.6.x

Convert set literal to tuple.

Resolves: #621