ARM-DOE/pyart #624: NEXRAD reflectivity map blank

troyofathens, 2016-11-25:
I'm very new to this module so forgive me if I ended up missing something but I downloaded Level 2 NEXRAD data directly off of NCDC I then downloaded one of the files as is into a directory and copied your example from pyart/examples/plotting/plot_nexrad_reflectivity.py word for word but replaced your filename with the appropriate one that I downloaded. This resulted in a graph appearing with nothing on it, no title, no axes titles, nada.
[image]

scollis, 2016-11-25:
plot_ppi 

-sent from a mobile device-

troyofathens, 2016-11-25:
not entirely sure what you mean

scollis, 2016-11-25:
Don't use .plot use .plot_ppi 

-sent from a mobile device-

troyofathens, 2016-11-25:
now tried downloading the file differently so now it is in ar2v format according to ncdc same result still is occurring
[image]

troyofathens, 2016-11-25:
found a temporary fix but does not look ideal. Using code that would display on a mercator plot like so...
[image]
resulted in the following...
[image]

scollis, 2016-11-27:
@troyofathens sorry need to use the map method Take  a look here:
https://github.com/scollis/radar_in_the_cloud/blob/master/notebooks/any_nexrad.ipynb

This is my code for reading from Amazon S3..

Key lines:
`base_date = "20161030_143011"
fmt = '%Y%m%d_%H%M%S' 

b_d =  datetime.strptime(base_date, fmt)

my_radar = get_radar_from_aws('KENX',b_d ) #replace with reading from your file
max_lat = 45
min_lat = 41
min_lon = -77
max_lon = -73

lal = np.arange(min_lat, max_lat, .5)
lol = np.arange(min_lon, max_lon, .5)

display = pyart.graph.RadarMapDisplay(my_radar)
fig = plt.figure(figsize = [10,8])
display.plot_ppi_map('reflectivity', sweep = 0, resolution = 'i',
                    vmin = -8, vmax = 64, mask_outside = False,
                    cmap = pyart.graph.cm.NWSRef,
                    min_lat = min_lat, min_lon = min_lon,
                    max_lat = max_lat, max_lon = max_lon,
                    lat_lines = lal, lon_lines = lol)
`

scollis, 2016-11-27:
[image]

jjhelmus, 2016-11-28:
@troyofathens Add the `fig = plt.figure(figsize=(6,5))` line to same Jupyter notebook cell where the plotting is done with the **display** object.  Alternatively you can remove all the `plt`, `fig` and `ax` lines/arguments.  

Either of the following should work if put them in their own cells:

``` Python
fig = plt.figure()
ax = fig.add_subplot(111)
display.plot('reflectivity', 0, title='NEXRAD Reflectivity',
             vmin=-32, vmax=64, colorbar_label='', ax=ax)
display.plot_range_ring(radar.range['data'][-1]/1000., ax=ax)
display.set_limits(xlim=(-500, 500), ylim=(-500, 500), ax=ax)
plt.show()
```

``` Python
display.plot('reflectivity', 0, title='NEXRAD Reflectivity',
             vmin=-32, vmax=64, colorbar_label='')
display.plot_range_ring(radar.range['data'][-1]/1000.)
display.set_limits(xlim=(-500, 500), ylim=(-500, 500))
```

With the `%matplotlib inline` command, figures are rendered at the end of the cell where they were created.  Any updates to the figure in other cells does not affect the rendered plot.

jjhelmus, 2017-01-09:
@troyofathens Did this fix your issue?  Can this be closed?

jjhelmus, 2017-02-02:
closing, feel free to re-open if this issue remains.

closed 2017-02-02 by jjhelmus