ARM-DOE/pyart #628: Some consideration about map_gates_to_grid on a single radar case.

striges, 2016-12-24:
I compared `map_gates_to_grid` results with some other grid results using the Weather Climate Toolkit, WDSS-II, and Radx. I am doing so as I am trying to porting the 'map_gates_to_grid` method to GPU as this one suits SIMD architecture better than many other interpolation ways. But I think some improvements are needed. 

The case I am showing here is KDOX on 10/29/2012 near 0005UTC. The domain size is -230km~230km with 0.5km resolution, altitude at 3500m

map_gates_to_grid
[image]

map_to_grid
[image]

Radx (Cartesian Interpolation)
[image]

The weather climate toolkit (IDW Interpolation)
[image]

There are three major differences between pyart results and Radx/WCT results:
1. Basically, `map_gates_to_grid` created a larger mosaic than Radx and WCT, actually pixels at out extent  is below 0 deg elevations scan, simply because the 3D buffer zone of a gate with RoI extends under the lowest elevation scan. I think this can be easily fixed by applying as hard mask there to exclude such zone.

2. `map_gates_to_grid` produces very sharp edge of "bull-eye" like nearly radar. Well, there are bright-bands in this case, just over the radar at height about 3.5km. However, if we carefully observe those rings, there are blank pixels, simply because those pixels are. Using horizontal blurring to fill these blank pixels actually is a bad idea, because it will mess up the convective cloud structures if we check the RHI, (see V. Lak's paper in 2005). This indicated an extreme corner case here: I can specify a grid centered at a point with no gates' RoI can reach it, but with a very large size (e.g. 460km cell size), this grid should cover many gates but `map_gates_to_grid` will produce a empty pixel! I propose the condition of inclusion to `dist - roi >= cell_size`, which roughly means if a pixel is touched with RoI buffer zone, it is affected.

3. Texture pattern. We can easily observe the texture of divergent rays in first two pictures, especially when gates is away from radar. And such problem will not ease if we apply a lowpass filter. Radx and WCT produces ring-like texture pattern. However, in radx or wct's results, a lowpass filter can remove any texture pattern there. The reason is how  number of gates affected pixels, I drew a figure of how many gates are affect each pixels in this case. Here is the result:
[image] 
(Well this figure is plotted after proposed changed in last point, this is not important). We can obviously see why such textures comes out. My idea is, instead of collecting all gates affecting on a pixel, we only collected the nearest 8~n (consider 8 vertex in a cube, but we need more to make smoothed result) gates, and we need prevent RoI propagates too wide to other rays, which I haven't figure out a good idea. BTW, another indirect result is some convective core is got removed due to averaging in a too large area. Another side issue is a single, standalone, dangle pixel will produce a circle, this is definitely wrong, because we in fact extrapolating pixels using interpolations. I think removing those pixels should be in a quality control stage, good QCed data should not have such issue, but it will be good to address these issue out.

In summary, `map_gates_to_grid` is an excellent algorithm that fits SIMD processing perfectly. I made a CUDA porting test and it can grid this case in 0.02sec. This is an amazing speed. But my suggestions is we need carefully consider how gates are affecting pixels, a buffer zone is too simple. I don't think Radx or WCT results are good either, lowpass postprocessing is bad idea but their results cannot be used without it. But simple interpolations can avoid all those cases above.

striges, 2016-12-29:
OK. I got some updates here. 
1. A hard cut below 0.5(actually, 0) is very easy. We can easily reverse calculate (x, y, z) to (ele, az gate) assuming radar is at (0,0,0) for each grid cell. Then we mask all gates whose elevations are below 0 or above 20 (or whatever reasonable maximum)

For 2 and 3, I simply propose replace RoI function as following: a gate has 2 x beam_width (or 3 x beam_width at high elevations) angular affects (in both elevation and azimuth, like a cone) and 2 * gate_size radial affects, The weight function is `0.005^(angular_diff) / dist^2`. This method is a combination of MRMS's method and IDW weighting, somehow producing higher results than n-nearest neighbor interpolation, but avoid any searches at all, keep the advantage of original mapping function.

So here is my preliminary result. 
[image]

swnesbitt, 2016-12-29:
Good stuff - a couple of comments:

1) I would be careful about the lower and upper elevation angle limits -
maybe make these an input parameter to the function?  If one is not using
NEXRAD data there could be a wide variety of scanning strategies (including
negative elevation scans from mountain-top radars, and VCPs from research
radars with interleaved scans with different elevations).  Don't want to
introduce "pulsing" in the echo statistics in this case.

2) Clearly the bright band is a remaining issue here.  Could there be an
algorithm that has a more horizontal ROI and in (and near) specified bright
band layer to get rid of the rings?  Obviously this would work better in a
warm season environment vs a cool season environment?  Also, there could be
some sort of rho_hv melting layer detection in the function as well (but
would work less well with range) - have to think about that a bit more.

Keep up the good work!
-Steve

striges, 2016-12-29:
I updated pictures in original cases, made it clear to show low values introduced in original methods.
@swnesbitt. Set a parameter is not a problem at all, I'd like to just point out we need a elevation cut, preventing "extrapolating data" using "interpolation". Ring issue, I have been bothered by this so long. I talked with so many folks in AMS, AMS tropical, AMS radar. In Zhang (also with Lak???) (2005)'s paper, horizontal blurring over rings is really a bad idea in convections. You can easily see how cloud structure is messed up in 3D. I originally thought a pure 3D weighting is the best idea, like currently pyart does, but results are not good at all. 
The best way is to reduce bright bands value in post-processing, just like MRMS does. However, the problem is, without isothermal reflectivity at 0, -5 (we need external temperature inputs), it is difficult to reduce them.
I am currently working with some computing folks if we can get some ideas from deep learning and computer vision to *visually* get a better results.

striges, 2016-12-29:
Sorry I don't have time to create a pull request. Here is my (CUDA) C code to implement them, hoping it will be helpful for another contributor to wrap it. 
https://gist.github.com/striges/7888203ea5cd7ce1e19445bfe78c4d87

scollis, 2017-03-21:
Just "Re-Pinging" Anyone out there available to write up an implementation of this?

scollis, 2017-07-13:
This is all great info and will be preserved here. We really need a separate repo for cases and studies et al. However, in terms of an issue I am closing this. Thanks all for the good discussion.

closed 2017-07-13 by scollis