ARM-DOE/pyart #636: Python 3.6 Incompatibility with pyart.correct.despeckle_field

vlouf, 2017-03-01:
While running pyart on python 3.6 I encountered this problem:

```
pyart/correct/despeckle.py in _adjust_for_periodic_boundary(data)
    199     i1 = 0
    200     i2 = np.shape(labels)[0] / 2
--> 201     old_labs = np.unique(labels[i2][labels[i2] > 0])
    202     for i, lab in enumerate(old_labs):
    203         cond = labels == lab

IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices
```

While on python 3.5:

```
pyart/correct/despeckle.py:201: VisibleDeprecationWarning: using a non-integer number instead of an integer will result in an error in the future
  old_labs = np.unique(labels[i2][labels[i2] > 0])
pyart/correct/despeckle.py:204: VisibleDeprecationWarning: using a non-integer number instead of an integer will result in an error in the future
  indices = np.where(labels[i2] == lab)
pyart/correct/despeckle.py:207: VisibleDeprecationWarning: using a non-integer number instead of an integer will result in an error in the future
  labels = labels[0:i2]
```

This problem is solved by changing line 200 of pyart/correct/despeckle.py to:
` i2 = int(np.shape(labels)[0] / 2)`

jjhelmus, 2017-03-01:
This is not really an issue with Python 3.6 but rather that indexing using floats is no longer allowed in NumPy 1.12 which is what was likely installed with Python 3.6.  The recommended fix is fine.  Thanks for reporting this and the suggestion @vlouf.

jjhelmus, 2017-03-01:
closed by 3d3ab692cdd270a1

closed 2017-03-01 by jjhelmus

CSlocumWX, 2017-11-18:
While @tjlang's fix works, I believe that this line should be changed so that it is [PEP 238](https://www.python.org/dev/peps/pep-0238/). With version Python 2.2, the integer division operator, '//', was added. Possibly consider changing the line from
`i2 = int(np.shape(labels)[0] / 2)`
to
`i2 = labels.shape[0] // 2`

tjlang, 2017-11-19:
Updated https://github.com/ARM-DOE/pyart/pull/674 to address this issue.