ARM-DOE/pyart #641: Moving Example Data to a permanent location

Rumpkie, 2017-03-13:
We should move the example data from: https://engineering.arm.gov/~jhelmus/pyart_example_data/
to a permanent location.

zssherman, 2018-04-04:
@scollis For this, should we have a data folder under examples with all the files? Or maybe a separate data repository. The first option, the files would get downloaded when the pyart repository is cloned, and not sure if that option is desired, seeing as the files are currently separate at the engineering jhelmus link.

kmuehlbauer, 2018-04-04:
@zssherman A separate repo would indeed be a good solution. Most obvious reason is to not clutter the source code repo with huge data. It works quite well in other packages I maintain. There are different approaches for retrieval though.

zssherman, 2018-04-04:
@kmuehlbauer Very good point, I like that idea. Yeah a separate repo seems ideal.

mgrover1, 2022-03-09:
Now that we have this module/in a permanent location, we can close!

closed 2022-03-09 by mgrover1