ARM-DOE/pyart #646: ImportError: No module named io

ucrkarthik, 2017-04-10:
**Compiled the source code:** 
[kvenkatesa@nid00030 pyart]$ python setup.py build
running build
running build_py
running build_ext

**Installed in my home directory:**
[kvenkatesa@nid00030 pyart]$ python setup.py install --user
running install
running bdist_egg
running egg_info
writing pyart.egg-info/PKG-INFO
writing top-level names to pyart.egg-info/top_level.txt
writing dependency_links to pyart.egg-info/dependency_links.txt
reading manifest file 'pyart.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no directories found matching 'docs'
no previously-included directories found matching 'docs/build'
no previously-included directories found matching 'docs/html'
warning: no previously-included files matching '*.pyd' found anywhere in distribution
warning: no previously-included files matching '.git*' found anywhere in distribution
warning: no previously-included files matching '.DS_Store' found anywhere in distribution
warning: no previously-included files matching '.mailmap' found anywhere in distribution
writing manifest file 'pyart.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/pyart
copying build/lib.linux-x86_64-2.7/pyart/tree.so -> build/bdist.linux-x86_64/egg/pyart
copying build/lib.linux-x86_64-2.7/pyart/__init__.py -> build/bdist.linux-x86_64/egg/pyart
creating build/bdist.linux-x86_64/egg/pyart/tests
copying build/lib.linux-x86_64-2.7/pyart/tests/__init__.py -> build/bdist.linux-x86_64/egg/pyart/tests
copying build/lib.linux-x86_64-2.7/pyart/tests/base.py -> build/bdist.linux-x86_64/egg/pyart/tests
copying build/lib.linux-x86_64-2.7/pyart/tests/test_tree.py -> build/bdist.linux-x86_64/egg/pyart/tests
byte-compiling build/bdist.linux-x86_64/egg/pyart/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-x86_64/egg/pyart/tests/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-x86_64/egg/pyart/tests/base.py to base.pyc
byte-compiling build/bdist.linux-x86_64/egg/pyart/tests/test_tree.py to test_tree.pyc
creating stub loader for pyart/tree.so
byte-compiling build/bdist.linux-x86_64/egg/pyart/tree.py to tree.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying pyart.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyart.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyart.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyart.egg-info/not-zip-safe -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyart.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
creating 'dist/pyart-0.2.3-py2.7-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing pyart-0.2.3-py2.7-linux-x86_64.egg
removing '/home/users/kvenkatesa/.local/lib/python2.7/site-packages/pyart-0.2.3-py2.7-linux-x86_64.egg' (and everything under it)
error removing /home/users/kvenkatesa/.local/lib/python2.7/site-packages/pyart-0.2.3-py2.7-linux-x86_64.egg: /home/users/kvenkatesa/.local/lib/python2.7/site-packages/pyart-0.2.3-py2.7-linux-x86_64.egg/pyart: Directory not empty
error removing /home/users/kvenkatesa/.local/lib/python2.7/site-packages/pyart-0.2.3-py2.7-linux-x86_64.egg: /home/users/kvenkatesa/.local/lib/python2.7/site-packages/pyart-0.2.3-py2.7-linux-x86_64.egg: Directory not empty
Extracting pyart-0.2.3-py2.7-linux-x86_64.egg to /home/users/kvenkatesa/.local/lib/python2.7/site-packages
pyart 0.2.3 is already the active version in easy-install.pth

Installed /home/users/kvenkatesa/.local/lib/python2.7/site-packages/pyart-0.2.3-py2.7-linux-x86_64.egg
Processing dependencies for pyart==0.2.3
Finished processing dependencies for pyart==0.2.3

**I am getting the following error when I try to import pyart.io. Here is the error I get: No module named io**

[kvenkatesa@nid00030 pyart]$ python
Python 2.7.5 (default, Sep 15 2016, 22:37:39)
[GCC 4.8.5 20150623 (Red Hat 4.8.5-4)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyart
>>> import pyart.io
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named io
>>>

zssherman, 2017-04-11:
Hi ucrkarthik,

From what I'm seeing is that you have the wrong pyart. There seems to be a pyart - Adaptive Radix Tree 0.2.3 from https://github.com/blackwithwhite666/pyart.git that you have, but the pyart I assume you want is the ARM pyart from https://github.com/ARM-DOE/pyart.git.

Try:
```bash
git clone https://github.com/ARM-DOE/pyart.git
```
then your:
```bash
python setup.py build
python setup.py install --user
```

zssherman, 2017-04-11:
Hi @ucrkarthik ,

From what I'm seeing is that you have the wrong pyart. There seems to be a pyart - Adaptive Radix Tree 0.2.3 from https://github.com/blackwithwhite666/pyart.git that you have, but the pyart I assume you want is the ARM pyart from https://github.com/ARM-DOE/pyart.git.

I would rename or delete the pyart Adapative Radix Tree directory and then try:
```bash
git clone https://github.com/ARM-DOE/pyart.git
```
then your:
```bash
cd pyart
python setup.py build
python setup.py install --user
```

scollis, 2017-05-02:
@ucrkarthik Is this still an issue?

itsjms, 2017-05-12:
I have this same issue.
```
AttributeError: 'module' object has no attribute 'io'
```

Uninstalled from conda and tried installing from source. Still no luck.

ucrkarthik, 2017-05-12:
You are using the wrong package.

ucrkarthik, 2017-05-12:
Thx for your help. That worked.

itsjms, 2017-05-12:
How do you rename or delete the pyart Adapative Radix Tree directory then?

itsjms, 2017-05-12:
I tried `pip uninstall pyart` with no luck. Still same error.

zssherman, 2017-05-12:
@ucrkarthik glad to hear, and @itsjms if you installed pyart the adaptive tree from source using `python setup.py install`, `pip uninstall` will most likely fail because `python setup.py install` causes no metadata to be left behind to be used to determine what files were installed. If you install arm_pyart by using `conda install -c conda-forge arm_pyart` and then import pyart are you still getting the error?

itsjms, 2017-05-12:
Yup same error.

itsjms, 2017-05-12:
`from pyart import Tree` still works so its deff using the wrong version.

scollis, 2017-05-12:
If you Change your conda env does that work
And what is the output of : which python ?

-sent from a mobile device-

zssherman, 2017-05-12:
Hmm, I'm curious, because the pyart adaptive tree I don't think is available through conda but the arm_pyart is, I would think that using python that was downloaded with the anaconda package would then take the conda arm_pyart when imported.

zssherman, 2017-05-12:
@scollis Ah very good point.

itsjms, 2017-05-12:
Which python now gives me `/usr/local/bin/python` I had installed it via `brew`

scollis, 2017-05-12:
Oh.. you are not using anaconda then..

so the error is when you type

"from pyart import io" ?

can you:

import pyart

pyart.io.read()

?

Also what is output of pyart.__version__ ?

final question.. if you installed from source did you ensure you changed 
dir out of source dir..

itsjms, 2017-05-12:
So if I add `export PATH=~/anaconda/bin:$PATH` to my `.bash_profile` `which python` returns `/Users/itsjms/anaconda/bin/python`

itsjms, 2017-05-12:
`pyart.io.read()` returns `'module' object has no attribute 'io'` `pyart.__version__ is` '0.2.3'

scollis, 2017-05-12:
Oh dear.. that is a very very old version..

we are at 1.8.0 :)

Sorry.. remind me: did you install from source?

if you installed from conda can you

conda update -c conda-forge arm_pyart

**USE WITH CARE**
if that fails you can go in and delete pyart from its location..
eg

print(pyart.__file__)

returns

/Users/scollis/anaconda/envs/houston/lib/python3.5/site-packages/pyart/__init__.py

so I could
rm -rf /Users/scollis/anaconda/envs/houston/lib/python3.5/site-packages/pyart/

and I know then I have zero trace left of my old pyart

itsjms, 2017-05-12:
Wonderful. `print(pyart.__file__)` was the golden ticket. At least now I have the wrong package of pyart gone (the one from https://github.com/blackwithwhite666/pyart.git) still have `arm_pyart` installed via anaconda but its not reading it yet...

itsjms, 2017-05-12:
I'm now getting `ImportError: No module named pyart` error. Maybe my path settings are wrong...

zssherman, 2017-05-12:
It seems that when you open python its using the brew installed python and then uses the pip packages, The usr path you provided seems to be the homebrew path. In the bash rc file as you mentioned earlier, the Homebrew PATH modification line above the Anaconda one. That way, Anaconda's  python, will take precedence, then close and reopen your terminal.

zssherman, 2017-05-12:
If you want to keep the homebrew path still and the other pyart is gone, you can also `pip install arm_pyart`.

itsjms, 2017-05-12:
`pip install arm_pyart` 👍  having other dependency issues now but pyart is working. If your having this issue it is definitely because you have the wrong version of pyart installed (https://github.com/blackwithwhite666/pyart.git) vs. arm_pyart

zssherman, 2017-05-12:
Awesome, yeah once the dependencies are installed, you should be good to go.

scollis, 2017-06-19:
Closing.. please reopen if issue persists

closed 2017-06-19 by scollis