ARM-DOE/pyart #655: Issues with some Sigmet IRIS file fields

tanelv, 2017-05-23:
I found an issue that I think has not been discussed before.
I am trying to read IRIS HydroClass data (radar.fields['radar_echo_classification']['data']), but for some reason the data is being 'compressed' to a shorter range (2x shorter than it should be). The matrix itself is the same size as other radar fields, but the further part is just empty (see figures below).

I checked other radar fields and all else seems to be fine, only radar_echo_classification is affected by this behaviour.

I checked it also in the IRIS Analysis software and there it shows as it should be, so I guess there is an issue somewhere in reading this radar field to Py-ART, which I could not figure out.

Below is a b-scan plot (upper figure is reflectivity and directly below is the 'compressed' HydroClass data).
The lowest figure is the HydroClass data in Cartesian, plotted in IRIS Analysis.

I included the original IRIS RAW file in the end of the message which was used so you could check it.

[image]
[image]

[SUR160603110001.RAW1GZR.zip](https://github.com/ARM-DOE/pyart/files/1021465/SUR160603110001.RAW1GZR.zip)

scollis, 2017-06-19:
Sorry for not getting back to you.. @zssherman Can you do a quick first check an see what is going on here.. @josephhardinee you would be the expert on this page when it comes to the format (and @jjhelmus ) any ideas? Does IRIS have more than one gate spacing in a file?

zssherman, 2017-06-20:
Hi @scollis, I'll take a look and let you know soon.

josephhardinee, 2017-06-20:
@scollis I believe the gate spacing is fixed in a single volume as it's set in the ingest_header, which is only included once based on memory and a quick glance at the docs. There is an option to do variable range bin spacing, but I'm not sure the details of how that's implemented (See the task_range_info structure). 

My guess just from glancing at this is that if you compare the values of hydroclass in the cartesian vs pyart they won't be the same. I'd hazard a guess that they're being read in as 16 bit when they are only 8 or something similar, and the rest of the range bins are getting filled in. 

That's just a first pass guess though. I'd start by looking at what the code for the product type is (should be 55 or 56) to determine whether it's one or two byte version of HID.

jjhelmus, 2017-06-21:
@tanelv What version of Py-ART are you using?  Sigmet 1-bytes HCLASS data was not supported until version 1.6.0 or after commit 34468971ba97eb7499e2739ad33bd5ec98d3999e was merged.  Before that you should have seen a warning about the read returning the raw unprocessed data.

tanelv, 2017-06-26:
I was using Py-ART version 1.6.0, that's where the issue was buried. As simple as that. I tested it with a newer version of Py-ART and did not get any compression problems anymore.

Thank you everyone!

scollis, 2017-06-26:
Thanks @tanelv . Closing

closed 2017-06-26 by scollis