ARM-DOE/pyart #657: Drop milliseconds from plot titles

vlouf, 2017-06-07:
Time with milliseconds is a bit overkill on plot titles, and takes extra space too. I propose these changes to drop milliseconds from plot titles:

`return num2date(times, units, calendar)` by `return num2date(times, units, calendar).replace(microsecond=0)` in functions `generate_radar_time_begin` (l126) and `generate_grid_time_begin` (l137) in `pyart/graph/common.py`

kmuehlbauer, 2017-06-07:
We could think about making this more general and introduce a user provided parameter format string to use (eg. `'%Y%m%d%H%M%S'`).  

@scollis Is there a mechanism within pyart for user preferences, which can be used to accomplish this?

scollis, 2017-06-07:
So the default config is here https://github.com/ARM-DOE/pyart/blob/master/pyart/default_config.py

At the moment the title is set here:
https://github.com/ARM-DOE/pyart/blob/master/pyart/graph/common.py#L228

It is iso format.. I think a PR that modifies this looking at the configuration would be welcome. I would suggest passing a strftime like argument.

At the moment I would suggest the community tackle this as it is in the category "It would be nice to have" We @Rumpkie @zssherman  would be happy to help.

vlouf, 2017-06-21:
Should this be a user prefenrece in the configuration file, or an arguments provided while plotting data, e.g.:

`gr = pyart.graph.RadarDisplay(radar)
gr.plot_ppi("reflectivity", dtime_format="%Y%m%d") 
`

closed 2018-05-22 by scollis