ARM-DOE/pyart #658: projection origin in mdv read and write

gamaanderson, 2017-06-12:
as pointed out by @TulipaSilva when writing grid to mdv and reading back to grid the projection origin may change
in **pyart/io/mdv_grid.py** line 125 reads:

    d["sensor_lon"] = grid.radar_longitude['data'][0]

while line 320 reads:

        origin_longitude['data'] = np.array(
            [mdv.master_header["sensor_lon"]], dtype='float64')

that means origin_longitude is over written by radar_longitude[0] (the same happens for lat and alt as well)

The main problem here is that Grid projection is global while MDV projection is field specific, so it is not clear how to combine the both. I would suggest to make it internally consistence, that means writing `origin_longitude` to `sensor_lon` even if it technicality incorrect. The other option would be to add an parameter in the `write_grid_mdv` function to regulate that.

as for @TulipaSilva there is no way (without changing pyart code) of correcting this after the mdv is written, so you should force it before: `grid.radar_longitude['data'][0] =  grid.origin_longitude['data'][0]`

scollis, 2017-06-12:
Thanks @gamaanderson 
I would be happy to see a PR along either of those lines. As far as I know only the Parana folks and @TulipaSilva are using this. The ARM team are all pretty busy on other things so this would need to be a community contribution.

TulipaSilva, 2017-06-13:
Thanks @gamaanderson  and @scollis, I'll try this solution. 
And, I'll also try to change the code for a PR, as soon as possible.
Thanks a lot.