ARM-DOE/pyart #659: Cannot open ARM netcdf file: AttributeError: 'numpy.str_' object has no attribute 'decode'

maahn, 2017-06-14:
Hi, I have an issue with reading ARM NSA KaSACR files with pyart 1.8.0, numpy 1.11.3 and python 3.5.3:

    import pyart
    fname = 'nsakasacrhsrhiC1.a1.20111208.212833.nc'
    radar = pyart.io.read(fname)

    ---------------------------------------------------------------------------
    AttributeError                            Traceback (most recent call last)
    <ipython-input-2-cef6d5ffac5b> in <module>()
          1 import pyart
          2 fname = 'nsakasacrhsrhiC1.a1.20111208.212833.nc'
    ----> 3 radar = pyart.io.read(fname)

    /Volumes/User/mmaahn/anaconda/envs/python3/lib/python3.5/site-packages/pyart/io/auto_read.py in read(filename, use_rsl, **kwargs)
        119         else:
        120             dset.close()
    --> 121             return read_cfradial(filename, **kwargs)    # CF/Radial
        122     if filetype == 'WSR88D':
        123         return read_nexrad_archive(filename, **kwargs)

    /Volumes/User/mmaahn/anaconda/envs/python3/lib/python3.5/site-packages/pyart/io/cfradial.py in read_cfradial(filename, field_names, additional_metadata, file_field_names, exclude_fields, delay_field_loading, **kwargs)
        190 
        191     # first sweep mode determines scan_type
    --> 192     mode = netCDF4.chartostring(sweep_mode['data'][0])[()].decode('utf-8')
        193 
        194     # options specified in the CF/Radial standard

    AttributeError: 'numpy.str_' object has no attribute 'decode'   

The error is not specific to a particular file, but I put one file here for testing
ftp://ftp.cdc.noaa.gov/Public/mmaahn/nsakasacrhsrhiC1.a1.20111208.212833.nc
(sorry, the file won't last long at that location)

Looks like it is related to this commit 81f47c00d6d640bbbc9ff8c4db1d632840a431dd. When I revert the commit, it's working again. @jjhelmus, do you remember the reason for that? In my understanding all python 3 strings are unicode, so the `.decode('utf-8')` should be not required? Btw, there is no error when using python 2.7.

scollis, 2017-06-14:
Thank @maahn 
This is high priority. I think we have seen this before.
@Rumpkie : Can you work on this.. feel free to use @zssherman

Rumpkie, 2017-06-14:
Yes @scollis I have time to look into this. I will get started on that this afternoon or tomorrow.
-Cory

Rumpkie, 2017-06-14:
@maahn what version of netcdf4 are you using?

maahn, 2017-06-14:
@Rumpkie I'm using 1.2.8, too. Thanks for looking into this. Max

vlouf, 2017-06-15:
Hi, 

I had this error too after updating my anaconda installation (python 3.6.1 and netcdf4 1.2.8).

I fixed it by changing line 192 in cfradial.py to:

`# first sweep mode determines scan_type
try:
    mode = netCDF4.chartostring(sweep_mode['data'][0])[()].decode('utf-8')
except AttributeError:
    mode = netCDF4.chartostring(sweep_mode['data'][0])[()]`

vlouf, 2017-06-15:
It's the pull request #656

zssherman, 2017-06-16:
Hi @vlouf, thank you for the pull request. Makes sense why it was failing because its trying to decode something that is already decoded. Cory is going to take a look.

scollis, 2017-06-19:
This should be fixed by  #656 Thanks @vlouf

closed 2017-06-19 by scollis