ARM-DOE/pyart #664: Travis failing

zssherman, 2017-06-21:
Travis is failing today even though the test past last night for a pending pull request. No new changes since, so possibly a travis issue https://blog.travis-ci.com/ or dependency issue. Appveyor does not fail.

This is a place holder, fix might conclude over time. Waiting to see.

kmuehlbauer, 2017-06-21:
@zssherman This is most probably a numpy 1.13 related issue. As `conda` picks up new `numpy 1.13` since some time between yesterday and today, the 💩 hit the fan.

The most bugs introduced through this are easy fixes (eg. indexing stuff, tolerance). But other changes in numpy revealed more problematic issues. I hope there aren't that much problems to resolve.

ping @scollis

kmuehlbauer, 2017-06-21:
py34 doesn't fail, because conda-forge dropped support. So there remained an older numpy version.

zssherman, 2017-06-21:
@kmuehlbauer Ah so it was a dependency issue. I was wondering why 3.4 was passing but not 2.7, 3.5 and 3.6. Makes sense, yeah hopefully not too many problems to solve as you said. Thank you for the information! It all make sense now. By the way, fantastic use of an emoji

scollis, 2017-06-21:
Thanks @kmuehlbauer ! You have saved us a lot of angst.. Amazed a minor version bump created so many issues.. Thank goodness for CI (Ping @jjhelmus !!)

kmuehlbauer, 2017-06-21:
@scollis 👍 Yes, at wradlib we are also struggeling with the aftermath of the numpy version change. 

But, numpy 1.12 was already complaining with several DeprecationWarnings. However, some issues came out of the blue. I'm thinking about incorporating `numpy` in the build matrix for better handling these cases.

zssherman, 2017-06-21:
@kmuehlbauer and @scollis I created two environments, both with a conda install c conda-forge arm_pyart. The only difference between these environments was numpy 1.13 in one and numpy 1.12 in the other.

radar.gate_z['data'] from numpy 1.12 environment
array([[  1.84995461e+01,   2.07109089e+01,   2.29296287e+01, ...,
          1.63470359e+04,   1.63626741e+04,   1.63783196e+04],
       [  1.78883737e+01,   2.00278340e+01,   2.21746514e+01, ...,
          1.62151672e+04,   1.62307339e+04,   1.62463080e+04],
       [  1.71753373e+01,   1.92309112e+01,   2.12938422e+01, ...,
          1.60613175e+04,   1.60768008e+04,   1.60922913e+04],
       ..., 
       [  7.13056802e+02,   7.96976848e+02,   8.80903421e+02, ...,
          1.64915334e+05,   1.65010864e+05,   1.65106399e+05],
       [  7.13056802e+02,   7.96976848e+02,   8.80903421e+02, ...,
          1.64915334e+05,   1.65010864e+05,   1.65106399e+05],
       [  7.13056802e+02,   7.96976848e+02,   8.80903421e+02, ...,
          1.64915334e+05,   1.65010864e+05,   1.65106399e+05]])

radar.gate_z['data'] from numpy 1.13 environment
array([[  1.80000000e+01,   2.00000000e+01,   2.30000000e+01, ...,
          1.63460000e+04,   1.63620000e+04,   1.63780000e+04],
       [  1.80000000e+01,   2.00000000e+01,   2.20000000e+01, ...,
          1.62150000e+04,   1.62310000e+04,   1.62460000e+04],
       [  1.70000000e+01,   1.90000000e+01,   2.10000000e+01, ...,
          1.60610000e+04,   1.60760000e+04,   1.60920000e+04],
       ..., 
       [  7.13000000e+02,   7.97000000e+02,   8.80000000e+02, ...,
          1.64915000e+05,   1.65011000e+05,   1.65106000e+05],
       [  7.13000000e+02,   7.97000000e+02,   8.80000000e+02, ...,
          1.64915000e+05,   1.65011000e+05,   1.65106000e+05],
       [  7.13000000e+02,   7.97000000e+02,   8.80000000e+02, ...,
          1.64915000e+05,   1.65011000e+05,   1.65106000e+05]]

Difference is the dtypes:
numpy 1.12
```python
>>>radar.gate_z['data'].dtype
float64```

numpy 1.13
```python
>>>radar.gate_z['data'].dtype
float32```

In our case travis test were failing when assert_allclose possible due to the atol=1e-14 not being correct for the new dtype change.

But in summary of what I currently have found, seems that these arrays for numpy default now to float32 instead of float64.

kmuehlbauer, 2017-06-22:
Thanks @zssherman for sharing. I haven't discovered no such change so far. Seems a bit strange.

scollis, 2017-06-23:
Fixed (?) by #665

closed 2017-06-23 by scollis