ARM-DOE/pyart #666: Check PEP8 compliance in CI

kmuehlbauer, 2017-06-29:
While it is wanted that every commit is checked for PEP8 compliance before submitting a PR, we all know that sometimes things slip under the radar. 

So I propose to add a PEP8 checker (eg. flake8 or similar) to the CI (travis) to check for PEP8 compliance directly after the install section and before testing section. With this applied every PR is tested and fails if the PEP8 compliance isn't met. Besides that, this lowers the burden for reviewers.

scollis, 2017-06-29:
We had exactly this discussion in the office on Tuesday.  Cory will be implementing this into the CI

dopplershift, 2017-06-29:
Personally, I recommend `pytest-flake8`, which is a plugin for pytest that will run flake8 on all the files as well as run tests.

scollis, 2017-06-29:
Thanks @dopplershift ! Have you implemented for MetPy? @Rumpkie will be doing this with help form @zssherman .. @zssherman  will be at Scipy and is keen to learn!

dopplershift, 2017-06-29:
Absolutely. We also have a bunch of flake8 plugins installed to catch various things, like print statements, check for copyright headers, etc. Works well and is easy to setup (and configure in `setup.cfg`):
1. `pip install pytest pytest-flake8 flake8-print flake8-copyright`  or conda
2. `pytest --flake8`
3. PROFIT!

Full list of plugins used are in [here](https://github.com/Unidata/MetPy/blob/master/environment.yml).

scollis, 2017-07-13:
Thanks @dopplershift . With @Rumpkie leaving Argonne I am assigning to Zach.

dopplershift, 2018-01-17:
It's usually pretty easy, and looking over your codebase, I don't think you'll have much problem because you're using a lot of the utility functions from numpy. The challenge happens when you move from nose's tools functions (like `assert_equal`) to use pytest's use of bare `assert` functions. You'll have to replace your use of `SkipTest` with something from pytest--I just don't remember what that would be.

zssherman, 2018-01-18:
Ah gotcha, well that is good to hear, and the functions like SkipTest etc, I'll look more in to. Thank you, that information helps alot!

PR #1060 (2022-03-09): Add new CI workflow to run tests
Closes #1035
Closes #666

This adds another GitHub action to run the ci and removes the old files in the `continuous_integration` directory
closed 2022-03-09 by mgrover1