ARM-DOE/pyart #669: gridded data doesn't make sense

Weiming-Hu, 2017-07-07:
Hi there,

**What I'm doing** I have just started exploring the power of `pyart`. I have a radar level II data file, and I would like to extract the gridded data and plot them. For example, I would like to extract data with altitudes of 10000m, 1000m, 100m, and 10m, and data within 0-10000m, 0-1000m, 0-100m, and 0-10m (with this, the output would be a collapsed 2D map from a 3D data cube).

**My code**
```
radar = pyart.io.read_nexrad_archive(data_file)
print('- max_elevation is set to %f m' %np.amax(radar.gate_altitude['data']))
# - max_elevation is set to 49489.528405 m

# map to grid
max_elevation = 1000
max_z = max_elevation
min_z = 0
grid_radar = pyart.map.grid_from_radars(
    radars=radar, grid_shape=(1, 100, 100),
    grid_limits=((min_z, max_z), (-300000.0, 300000.0), (-300000.0, 300000.0)),
    gridding_algo='map_gates_to_grid')

# plot
fig = plt.figure()
ax = fig.add_subplot(111)
ax.imshow(grid_radar.fields['reflectivity']['data'][0], origin='lower')
plt.title('elevation ' + str(min_z) + '~' + str(max_z))
plt.show()
```

**What's the problem** As shown below, maps on the first row make sense to me, because the higher plane we extract, the bigger the center circle blank area would be. However, why do the maps on the second row look similar/identical? Shouldn't they be a collection of data condensed/collapsed from data within their respective altitude range?

[image]

Thank you very much!

josephhardinee, 2017-07-07:
Weiming,
The gridding uses a region of influence to determine what data points get
included into each CAPPI point. Picture it as a little sphere around each
data point on your new grid that determines what is close enough to be
combined.This is not being changed in yours and so the only thing that
differs is what Z level you're calculating it at.
For your first request if you want a CAPPI at say 100m, set min_z=max_z=100
and the number of Z levels to 1.

For the second thing you're trying to accomplish (Projecting the 3d data
onto the ground plane), I would grid multiple levels with min_z=0,
max_z=1000 for instance, and then grid multiple levels (say 5 levels for
every 200meters) which is the first argument to grid shape. That gives you
a rectangular grid. Then pick what aggregation operator you would
like(mean, sum, etc) and apply it in the vertical direction.
The alternative is to set the Z region of influence to be very very
large(In this case 1km), and grid at the ground plane.

Weiming-Hu, 2017-07-07:
@josephhardinee Thank you! I think I understand now.

For my second case, if I understand it correctly, it will make little difference, only in terms of the output gridded data, between the two methods, `map_gates_to_grid` and `map_to_grid`, because the radius of influence of the grid point/radar data point is relatively small compared to the range of altitudes.

Please correct me if I'm wrong.

Thank you!

closed 2017-07-09 by Weiming-Hu