ARM-DOE/pyart #670: Add some support for X-Array in a staged manner.

scollis, 2017-07-13:
At the time of opening this issue I am at Scipy. One task I have/had at this meeting was to gauge the state of XArray. I have been using XArray for a separate project. It has very powerful indexing capabilities that could add a lot of functionality to radar grids.

The eventual goal is to switch the Py-ART grid model to Xarray. But as a transitional step I will create a pyart.map.grid_to_xarray function that will accept a Py-ART grid and return a X-Array dataset. 

One collateral impact will be this will naturally add cartopy support for grids. 

This functionally, at least initially will be optional and based on xarray being installed.

tjlang, 2017-07-13:
Did someone say cartopy support for grids? Y'all already got that. Here's a plot with multiple NEXRADs mosaicked via Py-ART, visualizing an ISS overpass of thunderstorms near San Antonio, Austin, and College Station (red stars) on 1/2/2017. Base map provided by cartopy and its StamenTerrain tiling. Was like two lines to get the radar on there, using Grid's method to get lat/lons, then a pcolormesh call.

[image]

scollis, 2017-07-13:
Yep.. But it is one line if you have it in an xarray container :) 
Sweet plot.

zssherman, 2020-03-26:
Xarray was added to grid routine, but not the radardisplay. Xarry was also used in the spectra object. I will leave this open for now .

mgrover1, 2022-04-12:
Coming back to this thread - we should like this to Discussion #1047

mgrover1, 2023-02-28:
# Requirements for Integration

As mentioned at previous meetings/discussions, we are moving forward with an xradar accessor, enabling us to use the full suite of functionality of Py-ART with xradar.

## Overall Goal
The idea here is add an accessor 

```python
xradar_tree.pyart.to_radar()
```

Which adds a collection of additional **methods**  (ex. `.get_gate_x_y_z`) to the xradar object that mimic the existing Py-ART radar object.

## Required Methods
The existing ```pyart.core.Radar()``` object has the following methods:
- [ ] ```__getstate__(self)```
  - Returns the state of the object, including the data
- [ ] ```__setstate__(self, state)```
  - Restore unpickleable entries from the pickled object
- [ ] ```init_rays_per_sweep(self)```
  - Initialize or reset the rays_per_sweep attribute
 - [ ] ```init_gate_x_y_z(self)```
  - Initialize or reset the gate_{x, y, z} attributes.
 - [ ] ```init_gate_longitude_latitude(self)```
  -  Initialize or reset the gate_longitude and gate_latitude attributes.
 - [ ] ```init_gate_altitude(self):```
  - Initialize the gate_altitude attribute.
 - [ ] ```_check_sweep_in_range(self, sweep)```
  - Check that a sweep number is in range.
 - [ ] ```check_field_exists(self, field_name)```
  - Check that a field exists in the fields dictionary.
        field_name : str, Name of field to check.
 - [ ] ```iter_start(self)```
  - """Return an iterator over the sweep start indices."""
 - [ ] ```iter_end(self)```
  - Return an iterator over the sweep end indices
 - [ ] ```iter_start_end(self)```
  - Return an iterator over the sweep start and end indices
 - [ ] ```iter_slice(self)```
  - Return an iterator over the sweep slice objects
   - [ ] ```iter_field(self, field_name)```
  - Return an iterator which returns sweep field data
 - [ ] ```iter_azimuth(self)```
  - Return an iterator which returns sweep azimuth data
 - [ ] ```iter_elevation```
  - Return an iterator which returns sweep elevation data
 - [ ] ```get_field(self, sweep, field_name, copy=False)```
  - Return the field data for a given sweep. Returns the numpy data array.
 - [ ] ```get_azimuth(self, sweep, copy=False)```
  - Return an array of azimuth angles for a given sweep. Returns a numpy array of azimuth angles.
  - [ ] ``` get_elevation(self, sweep, copy=False)``
  - Return an array of elevation angles for a given sweep. Returns a numpy array of elevation angles.
 - [ ] ```get_gate_x_y_z(self, sweep, edges=False, filter_transitions=False)```
  - Return the x, y and z gate locations in meters for a given sweep.
      With the default parameter this method returns the same data as
      contained in the gate_x, gate_y and gate_z attributes but this method
      performs the gate location calculations only for the specified sweep
      and therefore is more efficient than accessing this data through these
      attribute.
 - [ ] ```get_gate_area(self, sweep)```
  - Return the area of each gate in a sweep. Units of area will be the same as those of the range variable, squared.
 - [ ] ```get_gate_lat_lon_alt(self, sweep, reset_gate_coords=False, filter_transitions=False
    ):```
  - Return the longitude, latitude and altitude gate locations.
        Longitude and latitude are in degrees and altitude in meters.
        With the default parameter this method returns the same data as
        contained in the gate_latitude, gate_longitude and gate_altitude
        attributes but this method performs the gate location calculations
        only for the specified sweep and therefore is more efficient than
        accessing this data through these attribute. If coordinates have
        at all, please use the reset_gate_coords parameter.
 - [ ] ```get_nyquist_vel(self, sweep, check_uniform=True)```
  - Return the Nyquist velocity in meters per second for a given sweep.
 - [ ] ```info(self, level="standard", out=sys.stdout)```
  - Print information on radar.
 - [ ] ```_dic_info(self, attr, level, out, dic=None, ident_level=0)```
  - Print information on a dictionary attribute
  - [ ] ```add_field(self, field_name, dic, replace_existing=False)```
  - Add a field to the object.
 - [ ] ```add_filter(self, gatefilter, replace_existing=False, include_fields=None)```
  - Updates the radar object with an applied gatefilter provided
        by the user that masks values in fields within the radar object.
 - [ ] ```add_field_like(
        self, existing_field_name, field_name, data, replace_existing=False
    ):```
  - Add a field to the object with metadata from a existing field.
  - [ ] ```extract_sweeps(self, sweeps)```
  - Create a new radar contains only the data from select sweeps.
 - [ ] ```_gate_lon_lat_data_factory(radar, coordinate)```
  - Return a function which returns the geographic locations of gates
 - [ ] ```_gate_altitude_data_factory(radar)```
  - Return a function which returns the gate altitudes

zssherman, 2024-05-13:
@mgrover1 How were we in terms of this list?

PR #1622 (2024-08-29): ADD: Add to_radar() method to xradar objects
Add a to_radar() method to xradar using the accessor interface.

- [x] Closes #670 
- [x] Tests added
- [x] Documentation reflects changes
closed 2024-08-29 by mgrover1