ARM-DOE/pyart #677: DeprecationWarning::import imp module with Python 3.6

vlouf, 2017-08-02:
The following warning is raised with Python 3.6 while importing pyart:

`pyart/config.py:22: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses`

By following importlib documentation we can change l:78 of config.py from:

`cfile = imp.load_source('metadata_config', filename)`

to:

```cfile = importlib.util.spec_from_file_location('metadata_config', filename)
module = importlib.util.module_from_spec(cfile)
cfile.loader.exec_module(module)```

And not forgetting `import importlib.util`

It's only for Python 3.4 and newer. I don't know how to check python version for importing module, so that's why I did not commit a change.

zssherman, 2017-08-02:
Hi Valentin,

There is a open pull request #675 for the imp deprecation from yesterday. I changed a couple of files, and am opened for suggestions as it seems there are a couple of ways to change the way importlib is used to replace imp.

The changes do remove the pyart imp deprecation error, but sadly when importing pyart now, nose throws a deprecation error because the package nose has not been updated to fix there imp warnings.

scollis, 2018-02-13:
@zssherman is this still an issue?

zssherman, 2018-02-13:
@scollis This is related to pull request #675 the travis test need to run again.

zssherman, 2019-04-25:
Fixed, closing.

closed 2019-04-25 by zssherman