ARM-DOE/pyart #678: Script plot_nexrad_reflectivity.py hangs up upon execution

kenpryor67, 2017-08-03:
Hello All: I'm thrilled to have installed Py-ART on my Linux computer. The installation appears to have been successful. However, when I attempt to execute the script "plot_nexrad_reflectivity.py", the script hangs up with no apparent action (i.e. no image is displayed). No errors appear that indicate a problem with the code or installation. What might I need to do to get the script to execute correctly and produce the desired output as shown on the examples page (http://arm-doe.github.io/pyart/dev/auto_examples/plotting/plot_nexrad_reflectivity.html)? I'm currently running Python version 2.7.13 on this Linux system:
Linux rhw9150.star1.nesdis.noaa.gov 3.10.0-514.26.2.el7.x86_64 #1 SMP Tue Jul 4 15:04:05 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

kenpryor67, 2017-08-03:
This issue is fixed: I installed Py-Art in the Python3.6 environment and now works great, as shown by the attached NEXRAD image. 
[image]

scollis, 2017-08-03:
Great! Thanks @kenpryor67 ! Make sure you keep us updated on how you are using Py-ART.

closed 2017-08-03 by scollis