ARM-DOE/pyart #679: Bug when plotting two or more RadarMapDisplayCartopy.plot_ppi_map axes on same figure

tjlang, 2017-08-22:
The following code:
```
fig = plt.figure(figsize=(10, 10))
ax1 = fig.add_subplot(211)
dc1 = pyart.graph.RadarMapDisplay(rad8b)
dc1.plot_ppi_map('reflectivity', ax=ax1, fig=fig, resolution='i')
ax2 = fig.add_subplot(212)
dc2 = pyart.graph.RadarMapDisplay(rad8b)
dc2.plot_ppi_map('reflectivity', ax=ax2, fig=fig, alpha=0.05, resolution='i')
```
produces the following basemap plot, as expected:
[image]

However, when doing an analogous plot using RadarMapDisplayCartopy:
```
dc = pyart.graph.RadarMapDisplayCartopy(rad8b)
fig = plt.figure(figsize=(10, 10))
ax1 = fig.add_subplot(211)
dc.plot_ppi_map('reflectivity', ax=ax1, fig=fig, resolution='10m')
ax2 = fig.add_subplot(212)
dc.plot_ppi_map('reflectivity', ax=ax2, fig=fig, alpha=0.05, resolution='10m')
```
we get the following plot:
[image]

What's going on?

scollis, 2017-08-22:
> image

I think I know.

And it is a bug and a feature. Your axes have no projection so the code 
rejects your axes and puts in its own

https://github.com/ARM-DOE/pyart/blob/master/pyart/graph/radarmapdisplay_cartopy.py#L251

try:

projection = cartopy.crs.LambertConformal(
central_longitude=lon_0,
central_latitude=lat_0)
ax1 = plt.axes(projection=projection)

tjlang, 2017-08-23:
Thanks! Doing:
```
projection = cartopy.crs.LambertConformal(
    central_longitude=rad8b.longitude['data'][0],
    central_latitude=rad8b.latitude['data'][0])
dc = pyart.graph.RadarMapDisplayCartopy(rad8b)
fig = plt.figure(figsize=(10, 10))
ax1 = fig.add_subplot(211, projection=projection)
dc.plot_ppi_map('reflectivity', ax=ax1, fig=fig, resolution='10m')
ax2 = fig.add_subplot(212, projection=projection)
dc.plot_ppi_map('reflectivity', ax=ax2, fig=fig, alpha=0.05, resolution='10m')
```
produced:
[image]

closed 2017-08-23 by tjlang

amanii84, 2022-12-18:
there are no attribute called (pyart.graph.RadarMapDisplayCartopy) only  (pyart.graph.RadarMapDisplaybasemap)

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-9-deee3f0cb72e> in <module>
----> 1 dc = pyart.graph.RadarMapDisplayCartopy(radar)
      2 fig = plt.figure(figsize=(10, 10))
      3 ax1 = fig.add_subplot(211, projection=projection)
      4 dc.plot_ppi_map('reflectivity', ax=ax1, fig=fig, resolution='10m')

AttributeError: module 'pyart.graph' has no attribute 'RadarMapDisplayCartopy'
-----------------------------------------------------------------------------------------------------------
any help please

zssherman, 2022-12-18:
@amanii84  Module is RadarMapDisplay, the cartopy is no longer in the name as it's the main module radar map backend module.

amanii84, 2022-12-18:
@zssherman , please can you write it briefly because I do not understand you very well.

zssherman, 2022-12-18:
@amanii84 
dc = pyart.graph.RadarMapDisplay(radar)

is what it should be

amanii84, 2022-12-18:
where should I put cartopy to overlap the radar data on the orographic
sorry for this question, I am still a beginner in using radar data

zssherman, 2022-12-18:
@amanii84 
I would do something similar to this:
https://arm-doe.github.io/pyart/examples/plotting/plot_modify_colorbar.html#sphx-glr-examples-plotting-plot-modify-colorbar-py

And change the projection to the protection you want, and it should work.

zssherman, 2022-12-18:
The code is very similar, and where the projection is called, just switch out the projection

amanii84, 2022-12-18:
if you please, I want to make radar data in cartesian coordinates and transform it to NetCDF format.
could you help me?

zssherman, 2022-12-18:
I would move this to a discussion as this is a separate issue, but follow the example here, for converting to a Cartesian grid:
https://arm-doe.github.io/pyart/examples/mapping/plot_map_one_radar_to_grid.html#sphx-glr-examples-mapping-plot-map-one-radar-to-grid-py

then once you have a grid object do a:
pyart.io.write_grid('file paths, grid)

zssherman, 2022-12-18:
that will output to a Cartesian netcdf file.

amanii84, 2022-12-18:
thank you very mush, I will try all these scripts
I appretiate you for all these help.

zssherman, 2022-12-18:
No problem!

amanii84, 2023-01-03:
I Try all these scripts and I get the netcdf file but it is not in  Cartesian coordinate the lat , lon dimensions not found , the dimension is X, Y ,Z .
what I shoud do to get the lat, lon dimension.

zssherman, 2023-01-03:
@amanii84 the gridding functionality is to a Cartesian grid. If you want the lat lon preserved in this file you use:
write_point_lon_lat_alt=True in the grid writing function.

amanii84, 2023-01-17:
write_point_lon_lat_alt=True
it doesn't working

zssherman, 2023-01-17:
@amanii84 I would open a discussion here:
https://github.com/ARM-DOE/pyart/discussions
as this is outside the scope of this issue.
And we can discuss more there.