ARM-DOE/pyart #680: plot longitude or latitude slice given longitude and latitude coordinate pair

bugsuse, 2017-08-26:
Currently, it seems to just give a longitude and latitude point to plot a slice when i use plot_longitude_slice or plot_latitude_slice. Maybe I ignore some methods.

Given conditions, I want to plot a slice 

1) given a longitude and latitude point (such as lon = -90, lat = 36) and a angle to control the slice slope

2) given two longitude and latitude point (such as lon = [-90, -100], lat = [36, 40]), lon = -90 and lat = 36 means a longitude and latitude coordinate point, the rest of points means another coordinate point

 Thank you for your help and advice!

meteoswiss-mdr, 2017-09-04:
I implemented an initial prototype on that. Whenever you want I make a pull request.

bugsuse, 2017-09-05:
@meteoswiss-mdr Thank you very much!

closed 2018-03-22 by bugsuse

bugsuse, 2018-03-22:
Could you tell me which one method can achieve the plot on current version? Thank you very much! @meteoswiss-mdr

meteoswiss-mdr, 2018-03-22:
Hi @bugsuse,

The function is called plot_lanlon_level and it has been implemented in the MeteoSwiss version of Py-ART:
https://github.com/meteoswiss-mdr/pyart/blob/pyart-mch/pyart/graph/gridmapdisplay.py

It is not the official one. @scollis, since there is a demand for it maybe we can accelerate the transfer of this function?

bugsuse, 2018-03-22:
It's wonderful! Thanks a million! @meteoswiss-mdr