ARM-DOE/pyart #682: Add nearest neighbor and most common in ROI in map_gates_to_grid

swnesbitt, 2017-08-28:
It would be useful to have a nearest neighbor and most common value within the radius of influence in the gridding routines.  I have some idea how to do this in the python implementation (and have tested it, but it is sloooooow), but in the fast c-code code written by @jjhelmus - ummm, yeah.

Use case would be gridding hydrometeor identification from polar data (you don't do HID on gridded data - shame! shame!), or for tags related to use of particular rainfall algorithms applied at the polar level, etc.

meteoswiss-mdr, 2017-09-04:
I had exactly the same issue with the hydrometeor ID (and any discrete data actually) and also made a fast hack on the python implementation to use the nearest neighbour but I agree that in C would be faster (and slower to implement :)).

scollis, 2018-05-22:
Closing via #683 
Note help optimizing and speeding this up would be welcome

closed 2018-05-22 by scollis