ARM-DOE/pyart #684: Issues with the implementation of the Vulpani technique

scollis, 2017-09-04:
I had a brief discussion with Nitin Bharadwaj, who leads ARM's radar engineering efforts at the Radar Conference. 
He had Vulpani with him and it seems we have issues with the way in which the method was implemented. I am opening this issue as a placeholder.

https://github.com/ARM-DOE/pyart/blob/master/pyart/retrieve/kdp_proc.py#L742

scollis, 2017-09-18:
@nitinbharadwaj  gave further feedback and believes the implementation to have issues. @wolfidan any ideas? Did you test the code? I propose we make the method local until someone can do some more testing.

meteoswiss-mdr, 2017-12-04:
@scollis,

I have checked the code with the original version from Vulpiani (He kindly gave it to me). I can confirm that there was an error in the implementation. I have corrected it and it should be ready to pull. As soon as you tell me I will do so.

scollis, 2017-12-04:
Submit a PR at your convenience. I am on travel next week. I hope to dedicate some more time to Py-ART early next year as we cut a new release. Thanks for your work on this @meteoswiss-mdr

scollis, 2018-02-13:
Thanks @meteoswiss-mdr for your PR.. #707 closes this out.. If @nitinbharadwaj could do a test and confirm that will be great... Please reopen if issues persist

closed 2018-02-13 by scollis