ARM-DOE/pyart #686: Python 3.6 breaks arm_pyart conda package

swnesbitt, 2017-09-20:
Get a numpy error message on fresh anaconda 3.6 install using the conda-forge pyart package.  4 students in my class had this problem.

scollis, 2017-09-20:
Thanks Steve will follow up

-sent from a mobile device-

zssherman, 2017-09-20:
Hi @swnesbitt, are you having the issue during the installation process, or when you read a file with pyart.io.read? We ran into a similar issue before with the reading of files with pyart in later installs, if that is the issue.

swnesbitt, 2017-09-20:
we were trying to use pyart.io.read, although pyart.io.read_cfradial had
the same issue...

-- 
Steve Nesbitt, Associate Professor
Department of Atmospheric Sciences, University of Illinois at
Urbana-Champaign
Room 3060 Natural History Building, 1301 West Green Street, MC-104, Urbana,
IL 61801 USA
voice : +1.217.244.3740  fax : +1.217.244.1752
internet : snesbitt@illinois.edu ;
https://www.atmos.illinois.edu/people/snesbitt ;
http://publish.illinois.edu/snesbitt

zssherman, 2017-09-20:
Ah gotcha, I think I know what it is. If the error was ''numpy.str_' object has no attribute 'decode'" Valentin had a fix in #656 pretty much it is trying to decode a string already decoded in newer versions. I would try a python setup.py install of the current master of pyart or a pip install git+https://github.com/ARM-DOE/pyart.git. Yeah hopefully soon we will have a new release with conda.

swnesbitt, 2017-09-20:
Thanks.  Yes, we were able to install it from source.

scollis, 2017-09-20:
We need a new release soon.
Will look at doing that next week

-sent from a mobile device-

zssherman, 2018-03-01:
Fixed in [release 1.9.0 Picasso](https://github.com/ARM-DOE/pyart/releases/tag/v1.9.0-picasso). @scollis Want me to close?