ARM-DOE/pyart #690: Code Stalls When Using pyart.correct.dealias_fourdd

pjmarinescu, 2017-09-29:
Hello, 

I have been using pyart to read in and dealias both NEXRAD Level 2 and CSU-CHILL radar velocity data. I have been using the pyart.correct.dealias_fourdd function. It works great for the NEXRAD data, but the code stalls for the CSU-CHILL data. The CSU-CHILL data I am using is not in the traditional CHL format, but has been converted to uf files (these files are ~30 MB). 

From my own tests, I believe the code is stalling at the following line:
_rsl_interface._label_volume(rsl_volume, radar)

One difference between the data is that the CSU-CHILL data is sector scans, while the NEXRAD data is full sector PPIs. 

Do you have any ideas what may be causing the code to stall with the CSU-CHILL uf files?

I have included below the lines that read in the data, as well as the call to the dealias_fourdd.

file = '/Volumes/42/CHILL/ORIG/20160717/CHL20160717_222347.uf'
radar = pyart.io.read_uf(file,field_names={'DZ':'DBZ', 'VE':'VEL'})
dealias=pyart.correct.dealias_fourdd(radar,sounding_heights=aint, sounding_wind_speeds=wsint,\
                                      sounding_wind_direction=wdint,vel_field='VEL')

Please note that I have tried several different read commands (pyart.io.read_uf, pyart.io.read, not including the field_names specification) with the same code stall. Also, I use the same sounding data for the NEXRAD and CHILL data, and since it works with the NEXRAD data, I am assuming that is not the issue for the CHILL data.

Thanks for your time,

Peter

freemansw1, 2017-10-17:
Just to follow up on this- I've been working on debugging this now and have narrowed it down to the while loop in the findRay function of dealias_fourdd.c. It appears that under certain circumstances, the while loop can never find a ray that is sufficiently close, so it gets caught in an infinite loop. 

I have been able to somewhat resolve the problem by replacing the relatively complex findRay function with a much simpler version and just putting closeness checks into continuity_dealias. Working now on a sanity check before I do a PR, any thoughts on this? I don't see a reason to keep that while loop in the findRay function and not replace it with a for loop.

scollis, 2017-10-17:
Hey Sean,
Is this in Py-ART or 4DD itself?

<https://github.com/ARM-DOE/pyart/issues/690#issuecomment-337350736>, or mute
the thread

freemansw1, 2017-10-17:
It looks like it's in the 4DD C code itself- although in the Py-ART version of it (so the one in pyart/correct/src). I'm not familiar with the original 4DD code though, so this could be from the modifications made to the code to interface with python.

scollis, 2017-10-17:
Yep.. I would say put a PR against it.. We can contact the guys from UWash
(who use Py-ART anyway) so they can update 4DD.. CC-ing in Stacey Brodzik.

freemansw1, 2017-10-18:
Okay, I've put a PR on this, #693 . Not entirely sure that my way of fixing it is entirely correct/robust, but at the least, it avoids the infinite loop. It's still odd to me that these files have this problem, but I assume it is some sort of a sector scan issue or something that hasn't yet been encountered.

srbrodzik, 2017-10-18:
The original 4DD code was written in C a long ago by one of our grad students.  I have no idea who created the python version but I do know that it exists.  The analysis that freemansw1 did seems reasonable and I have no problem if he implements it.

scollis, 2018-02-13:
I think this is fixed by #693 Please reopen if still an issue

closed 2018-02-13 by scollis