ARM-DOE/pyart #694: Breaks when trying to read an ODIM HDF5 file with missing tilts.

vlouf, 2017-10-30:
We had a problem at the BOM with some ODIM HDF5 files. Some scan can be missing some data types.

For example, on one of the file, tilt 10 does not contain reflectivity.  The file itself is valid, but the volume scan does not contain all the data it normally would.

This can happen for lots of reasons but the most common one is that the communications line to the radar was running slowly so some of the moments were dropped.

It would be nice if the reading of ODIM HDF5 files was robust enough to still cope with files like this.  In general, it is not guaranteed that every tilt will always contain the same set of data types.

One solution could be to change l.323 of pyart/aux_io/odim_h5.py to:
```python
try:
    sweep_data = _get_odim_h5_sweep_data(fid[dset][h_field_key])
except Exception:
    sweep_data = np.zeros((rays_in_sweep, nbins))
    sweep_data = np.ma.masked_where(sweep_data == 0, sweep_data)
```

scollis, 2018-05-22:
@vlouf Are you able to put a sample file where this occurs some where? We can then ask the community for help

vlouf, 2018-05-23:
@scollis Attached is this file [71_20150420_175459.pvol.h5.zip](https://github.com/ARM-DOE/pyart/files/2029050/71_20150420_175459.pvol.h5.zip), that return this error:

```python
KeyError                                  Traceback (most recent call last)
<ipython-input-13-5508b39d3535> in <module>()
----> 1 radar = pyart.aux_io.read_odim_h5('71_20150420_175459.pvol.h5', file_field_names=True)

/miniconda3/envs/radar/lib/python3.6/site-packages/pyart/aux_io/odim_h5.py in read_odim_h5(filename, field_names, additional_metadata, file_field_names, exclude_fields, **kwargs)
    321         # loop over the sweeps, copy data into correct location in data array
    322         for dset, rays_in_sweep in zip(datasets, rays_per_sweep):
--> 323             sweep_data = _get_odim_h5_sweep_data(hfile[dset][h_field_key])
    324             sweep_nbins = sweep_data.shape[1]
    325             fdata[start:start + rays_in_sweep, :sweep_nbins] = sweep_data[:]

h5py/_objects.pyx in h5py._objects.with_phil.wrapper()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper()

/miniconda3/envs/radar/lib/python3.6/site-packages/h5py/_hl/group.py in __getitem__(self, name)
    165                 raise ValueError("Invalid HDF5 object reference")
    166         else:
--> 167             oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
    168 
    169         otype = h5i.get_type(oid)

h5py/_objects.pyx in h5py._objects.with_phil.wrapper()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper()

h5py/h5o.pyx in h5py.h5o.open()

KeyError: "Unable to open object (object 'data2' doesn't exist)"
```

vlouf, 2018-05-23:
However, I don't have any problem when I use `pyart.aux_io.read_radx`

vlouf, 2020-06-11:
It's an old issue but it became a problem for me again recently, because there are so many fields missing in the recent dual-pol files due to comms issues. The error is of this nature:
`KeyError: "Unable to open object (object 'data4' doesn't exist)"`

pyART assumes the same moments are available in every tilt.  That could lead to this problem.  There is no guarantee that the same moments are always produced for every sweep. 

Because the data does not exist, the solution is to put dummy fields where data is missing. I've fixed it by catching the error and create an empty slice full of nan. I'll submit a pull request soon.

scollis, 2020-06-11:
As always Thank you Valentin 

-sent from a mobile device-

zssherman, 2020-06-12:
Fixed in #938 closing.

closed 2020-06-12 by zssherman