ARM-DOE/pyart #696: The circular reference in Radar Class

hhuangwx, 2018-01-16:
The LazyLoadDict used in the Radar class caused a circular reference, which will make the memory used by the class never collected by the automatic gc system (known as the memory leakage).  
[image]

If you are agree with my opinion that the LazyLoadDict should not utilized, I can help to fix the above problem.

scollis, 2018-01-16:
Thanks @hhuangwx ! @zssherman take a look at this

scollis, 2018-01-16:
And yes.. a PR would be very welcome..

scollis, 2018-01-17:
Ha! Your PR found a bigger issue as well thanks to the CI. Numpy's np.fromstring is depricated. 
We need to fix this first. It will not impact CF-Radial.. but it will impact the binary readers.

zssherman, 2020-03-26:
Closing this for now.

closed 2020-03-26 by zssherman