ARM-DOE/pyart #699: Replace instances of np.fromstring with np.frombuffer

scollis, 2018-01-17:
This is poorly documented in Numpy but looking here:
https://docs.scipy.org/doc/numpy-1.14.0/reference/generated/numpy.fromstring.html

It looks like fromstring will soon no longer fall back to frombuffer. We need to look at changing all our np.fromstring calls to np.frombuffer

scollis, 2018-01-17:
actual message 
```
  sw_data = np.fromstring(decompr_data, np_form).astype('float32')
/Users/scollis/anaconda/envs/broke/lib/python3.5/site-packages/pyart/io/mdv_common.py:607: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
```

scollis, 2018-01-17:
@goharkey submitted a PR but it only addresses Nexrad.

We also want to check carefully frombuffer is unpacking correctly

Call me untrusting

dopplershift, 2018-01-17:
>We also want to check carefully frombuffer is unpacking correctly

>Call me untrusting

That's what the tests are for. 😁

scollis, 2018-01-17:
Exactly.. my initial explore with mdv format has me hopeful

scollis, 2018-01-17:
@zssherman we missed one
```
test_auto_read.test_autoread_sigmet ... /Users/scollis/sandbox/pyart/pyart/io/sigmet.py:141: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  sigmet_data, sigmet_metadata = sigmetfile.read_data(full_xhdr=full_xhdr)
/Users/scollis/sandbox/pyart/pyart/io/sigmet.py:141: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  sigmet_data, sigmet_metadata = sigmetfile.read_data(full_xhdr=full_xhdr)
ok
test
```

scollis, 2018-01-17:
now of course the big issue is with Basemap @WeatherGod 
We'll comment over there but it is throwing all these warnings as well.

zssherman, 2018-01-17:
@scollis that is the one that I had to delete the build and do the python setup.py install build-ext -i and redo the nosetests and it got fixed. The one I asked about on slack.

scollis, 2018-01-18:
Okay thanks to @zssherman and new contributor @goharkay this is fixed via #700 #702 and #704

closed 2018-01-18 by scollis