ARM-DOE/pyart #713: Create color blind friendly colormap from Cameron Homeyer.

scollis, 2018-02-22:
See Tweet
https://twitter.com/AtmoScientist/status/966720414381682688
[image]

scollis, 2018-02-22:
Assign @rcjackson
For some reason the system will not let me do this

dopplershift, 2018-02-22:
Is he a collaborator or team member on the repo?

scollis, 2018-02-22:
Yeah.. Think that is my issue..

dopplershift, 2018-02-22:
That's the sole reason why having a contributor with "read only" access makes sense on a public project.

scollis, 2018-02-22:
Done.

Scott Collis

<https://github.com/ARM-DOE/pyart/issues/713#issuecomment-367823078>, or mute
the thread

swnesbitt, 2018-02-22:
This is pretty similar to @tjlang 's rainbow color table...

[image]

scollis, 2018-02-23:
Agree @swnesbitt but it does not hurt to have more. I will create a meta-class of colorblind friendly colormaps. something like pyart.graph.cb_cm

scollis, 2018-02-23:
ping @dopplerchase

swnesbitt, 2018-02-23:
OK, I'm going to dig out the Zipser-Nesbitt color table from the IDL
archives...

scollis, 2018-02-23:
Please do @swnesbitt ! It is Ed I have foremost in my mind when advocating for accessable colormaps.

swnesbitt, 2018-02-23:
[image]

dopplerchase, 2018-02-23:
There are pre-made functions to investigate the different colormaps found here: [link](https://matplotlib.org/cmocean/colormaps_viscm.html#colormaps-viscm )

[image]

scollis, 2018-02-23:
very cool.. @rcjackson I wonder if this will help http://astrolitterbox.blogspot.com/2012/11/rewriting-idl-color-table-routines-in.html

tjlang, 2018-02-23:
Yeah, I remember now, I could get LangRainbow42 to be relatively decent for color blindness (not perfect by any means), but it was crap for B&W printing. It is by design not perceptually uniform, the yellow is a big breakpoint. That causes the eye to pick out likely convective areas, provided you pick the center point of your dBZ scale well. It was a good exercise for learning how to make color tables and add them to Py-ART. I have been shocked at how much others have used it.

scollis, 2018-02-23:
So for Scipy I will be instructing my whole team to stay away from the NWS color
tables. We get nailed for bad color tables when we present outside the radar community. LangRainbow12 is not perfect but it is far better.

dopplerchase, 2018-02-23:
I agree
[image]

kmuehlbauer, 2018-02-23:
@scollis This comes straight back to some discussion we had, I currently do not remember where. Short story, we used viridis for some reflectivity plots in a paper. We had to change it back to rainbow, because editor and reviewer forced us to. 

I'm all for special weather radar colorblind friendly and perceptually uniform colormaps.

But I would greatly appreciate it, if we can have a dedicated package for this, from where all can benefit.

I really like the ideas doing this under the @openradar umbrella.

scollis, 2018-02-23:
Good idea @kmuehlbauer I was thinking about doing two things:
1) Regardless of where the colormaps reside we make a set of notebooks with @dopplerchase 's plots and a set of clear recommendations. These should reside in @openradar 
2) BAMS inbox article on this topic. We need to be able to nail reviews that are stuck in the past. Having a citable reference will greatly help.

scollis, 2018-02-23:
By the way @kmuehlbauer I am vice chair of the AMS Radar Committee.. We can look at having a formal recommendation on best practices that will be adopted by the AMS..

kmuehlbauer, 2018-02-23:
@scollis Great, any way forward is good!

Short Abstract for ERAD still in the ballpark? WDYT?

rcjackson, 2018-02-23:
The colormap is no good for black and white, but otherwise it looks like it should work well for colorblindness.
[image]

CameronHomeyer, 2018-02-23:
Fantastic work Bobby! This clearly shows another problem with the original (a luminance issue at the green-yellow seam). Please tweak the luminance assignment a bit following changes I made this morning: https://twitter.com/AtmoScientist/status/967048352784318465

scollis, 2018-02-23:
@kmuehlbauer I have another topic in mind for my presentation but I would be 100% supportive of an abstract on this!!! Can we put in two? If no one else wants to lead the abstract I will.

@CameronHomeyer Welcome! This is a good platform to iterate.. Looking forward to seeing what @rcjackson comes up with on your tweak.

scollis, 2018-02-23:
From Twitter https://twitter.com/njgoldbaum/status/967105515955335168
IDL colormaps in Python through YT

scollis, 2018-02-28:
How we going with this @rcjackson ?

rcjackson, 2018-02-28:
I was able to get @CameronHomeyer's colormap to work in Python, but have not gotten past that point yet. Need to test out the yt colormaps to see if there is anything colorblind friendly in there. After that I suspect adding them into Py-ART will be easy. I can get around to this tomorrow.

rcjackson, 2018-02-28:
[image]
Seam is gone @CameronHomeyer

CameronHomeyer, 2018-02-28:
Looks great @rcjackson! Thanks again for making this work.

rcjackson, 2018-03-01:
[image]
Changed the transition and also (I was using 50 colors, seems like setting nc to 15 makes the seam even less visible).

scollis, 2018-03-01:
@stechma2 how does this look to you?

stechma2, 2018-03-01:
@scollis This looks excellent to me. I can easily pick out the transitions between each of the colors and subsequently any changes in pattern in the examples... @rcjackson can you show a sample reflectivity plot of a common feature (i.e., MCS, supercell, whatever) just for reference?

dopplerchase, 2018-03-01:
I second @stechma2's motion for an example for reference

scollis, 2018-03-01:
@rcjackson can you whip something up from a radar file?

kmuehlbauer, 2018-03-01:
:+1: This looks very promising with respect to smoothness. Need to see greyscale figures from it.

Would be nice, if the data file and the python code to plot would be made available somewhere. So we could test same data also with our favorites.

scollis, 2018-03-01:
This will be part of the ERAD presentation @kmuehlbauer

scollis, 2018-03-01:
We will have a set of test cases for visualization.

scollis, 2018-03-01:
Once we have this in an accepted PR I will open a new issue covering the larger issue.

kmuehlbauer, 2018-03-01:
@scollis Yes, for sure. Would be good to have a playground with different data and different moments on the way to the ERAD presentation.

rcjackson, 2018-03-01:
[image]

Here is an example.

kmuehlbauer, 2018-03-01:
@rcjackson Nice. Are the lower end values masked? I do not see darker blue values. Could you set vmin,vmax to 0 and 55? Or some other fitting values?

scollis, 2018-03-01:
Its operational data.. they are discarded by the signal processor

kmuehlbauer, 2018-03-01:
Maybe I'm missing something. Are the colors fitted to specific dBZ values? Or does the colormap use the dynamic range of the data? Just curious.

rcjackson, 2018-03-01:
[image]
vmin, vmax is 0 to 60

rcjackson, 2018-03-01:
Gonna put in a PR with a new cm_colorblind class that will start out with just this colormap and its reverse. With more discussion more maps will be added later.

CameronHomeyer, 2018-03-01:
@rcjackson Looks great! Hard to believe this came together so fast, but my compass is tuned to academic life :)

scollis, 2018-03-01:
Closing as addressed by #716 We will open a more wide ranging issue which will be aligned with our ERAD work which will look at each moment and building out a set of colormaps in cm_colorblind. Seconding @CameronHomeyer this is a phenomenal effort going from tweet to code in 7 days.

closed 2018-03-01 by scollis

rrniii, 2018-11-07:
Hey all,
Thanks for sorting this and @scollis for pushing the issue in a recent paper. We have just implemented it in our operational products for a new year-long project.

https://sci.ncas.ac.uk/raine/resources/view/9036