ARM-DOE/pyart #714: python 3.6 read_nexrad_archive

swnesbitt, 2018-02-22:
Getting a bazillion depreciation errors when trying to read a nexrad level2 file using `pyart.io.read_nexrad_archive(filename)`

```
/Users/snesbitt/anaconda/lib/python3.6/site-packages/pyart/io/nexrad_level2.py:619: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(buf[ptr2: ptr2 + ngates], '>u1')
```

swnesbitt, 2018-02-22:
obvs can do this to hide 😄 
```
import warnings
warnings.filterwarnings('ignore')
```

scollis, 2018-02-22:
Update time for you. We fixed this in #700 
Py-ART 1.9.0 on conda-forge has this fixed.

scollis, 2018-02-22:
print(pyart.__version__)

zssherman, 2018-02-22:
@scollis beat me to it.

scollis, 2018-02-22:
I love answering the easy ones.

scollis, 2018-02-22:
@swnesbitt Let me know if this works so I can close the issue.

swnesbitt, 2018-02-22:
```
print(pyart.version)
<module 'pyart.version' from '/Users/snesbitt/anaconda/lib/python3.6/site-packages/pyart/version.py'>
```

scollis, 2018-02-22:
Yeah.. sorry 
`pyart.__version__`

swnesbitt, 2018-02-22:
'1.9.0.dev+35d6f51'

scollis, 2018-02-22:
Hmm ok .. That is odd then.. did you install from source or conda-forge?

swnesbitt, 2018-02-22:
source

swnesbitt, 2018-02-22:
trying conda-forge

swnesbitt, 2018-02-22:
that fixed it 👍

closed 2018-02-22 by swnesbitt