ARM-DOE/pyart #718: Improving plot_ppi_section for unordered VCPs (e.g., 88D SAILS)

deeplycloudy, 2018-03-12:
When WSR-88D data are in a SAILS VCP mode (elevation pattern below), it causes the PPI cross section plot to drop or oddly overplot some rays. This can be worked around with a manual call to `extract_sweeps`, but I propose that this situation be automatically detected and corrected for the user.

The list of elevations angles is: 
`[0.48339844, 0.48339844, 0.87890625, 0.87890625, 1.31835938, 1.31835938, 0.48339844, 0.48339844, 1.80175781, 2.41699219, 3.12011719, 0.48339844, 0.48339844, 3.99902344, 5.09765625, 6.41601562, 7.99804688, 0.48339844, 0.48339844, 10.01953125, 12.48046875, 15.60058594, 19.51171875]`

Before `extract_sweeps`
[image]

After `extract_sweeps`
[image]

Thanks to Jessica S., and undergraduate working with me, for pointing out this bug and finding the workaround.

scollis, 2018-03-12:
Thanks @deeplycloudy and Jessica! A PR on this would be most welcome.. Suggest having a method that can be called by the user and then called automatically by plot ppi xsects.. We have some travel coming up but can help in a few weeks

deeplycloudy, 2018-12-07:
I finally had time to write up a draft of this. I [put it in a gist](https://gist.github.com/deeplycloudy/d5d4f137dd7496434e09f1fbc2122b0f) for now.

scollis, 2018-12-07:
Fantastic.. I’m at AGU this week and Zach is busy with his AMS talk, but we will get to this soon. Looks fairly straightforward.. we will want to design some tests 

Thanks Eric!

-sent from a mobile device-