ARM-DOE/pyart #719: Mapping of satellite borne Doppler radar data

winash12, 2018-03-18:
Based on my question in the mailing list and this reference - https://journals.ametsoc.org/doi/pdf/10.1175/1520-0426%281994%29011%3C0572%3AMOADRD%3E2.0.CO%3B2 I do not understand why mapping (or interpolating) satellite borne Doppler radar data(for e.g. GPM precipitation radar) into a Cartesian coordinate system would not be part of PyArt. NCAR currently has two tools for this - REORDER and SPRINT but they are not being maintained right now. Any interest in adding this functionality to PyArt ?

winash12, 2018-03-20:
@scollis Thanks for adding this as an enhancement. Please correct me if I am wrong but the functionality /implementation code required for GPM should be a subset of the airborne borne Doppler radar given that the satellite moves in a constant trajectory around the earth unlike an aeroplane. A plane conceivably could move in a trajectory that would involve changes to all  three Eulerian angles but not a satellite.

swnesbitt, 2018-03-20:
There is open source IDL code for this purpose https://opensource.gsfc.nasa.gov/projects/GPM/index.php#software

Turns out to be more difficult to do this in practice than you might think, because the satellite geolocation is typically uncertain by as much as a few kilometers in the case of TRMM due to the failure of the star tracker on board. GPM is probably better, but may be several hundred meters off.  This was solved in TRMM by minimizing the differences in reflectivity by shifting the position of the satellite data around until it matches best, then proceed with the colocation. See work by Bolen and Chandrasekar.

kmuehlbauer, 2018-03-20:
Just as a sidenote there is the paper of [Schwaller and Morris (2011)](http://journals.ametsoc.org/doi/pdf/10.1175/2010JTECHA1403.1) and the recent paper of [Warren et al. (2017)](https://journals.ametsoc.org/doi/pdf/10.1175/JTECH-D-17-0128.1) which handle 3D matching of ground based radar and space born radar. 

@scollis If there is interest in python code, there is [this repo](https://github.com/vlouf/matchproj-python) from @vlouf using pyart for ground radar reading. The georeferencing and matching from the latter paper is also [implemented in wradlib](http://wradlib.org/wradlib-docs/latest/notebooks/match3d/wradlib_match_workflow.html).

winash12, 2018-03-20:
I found another one here - https://github.com/vecoveco/gpm

scollis, 2018-03-20:
I should also add wradlib has some capabilities 

-sent from a mobile device-

vlouf, 2018-03-20:
If you want, I volunteer to rework my script (the current repo is not up-to-date with what we use at the BOM), to look like more something we could implement in pyart..

winash12, 2018-03-21:
I also volunteer my time to work with @vlouf and probably @kmuehlbauer. We could take the best approach of both repos and come up with a best solution for PyArt. At the moment I am a noob but I can come up to speed very quickly.

vlouf, 2018-03-21:
I thought about it and I think I can port the latest version of the matching code used at the BOM into PyART without too much work. However, it requires [pyproj](https://pypi.python.org/pypi/pyproj?) (and the PROJ4 library)... I am almost certain that basemap already requires it, thus it wouldn't be a problem. But if it isn't something installed by basemap, I suppose it could be an optional package loaded when the matching function is called.

I'll create a new repo for that development.

winash12, 2018-03-21:
@vlouf i have not tried running your code yet. Can it handle  stand alone SR data ? I am talking about situations like over the oceans or over the mountains where ground radars may or may not be present. Also please correct me if I am wrong - your code requires installation of both HDF5 and HDF4 ?

kmuehlbauer, 2018-03-21:
@vlouf @winash12 I can recommend using netcdf4 to read the gpm and trmm files. If you need some inspiration, have a look into wradlib read_gpm and read_trmm functions. This prevents using hdf4 for reading the trmm data.

winash12, 2018-03-21:
@kmuehlbauer We may need to take this offline but correct me if I am wrong. Since the GPM data that I download from NASA is in HDF5 you are suggesting I convert from HDF5 to netCDF4 before I call read_gpm ?

kmuehlbauer, 2018-03-21:
@winash12 reading gpm or trmm data is not the problem. But correct georeferencing is needed to get the xyz coordinates of the 3D SR bins. And of course what projection to use.

kmuehlbauer, 2018-03-21:
No, netcdf4 is capable of reading hdf. No need to convert.

winash12, 2018-03-21:
I would be hoping to use the lat lon grid or the cylindrical equidistant projection but of course others may have other projections in which case we do need pyproj and proj.

winash12, 2018-03-21:
@kmuehlbauer if I understood you correctly one needs to go from radar coordinate system to Cartesian and then you do a projection from that is that correct?

kmuehlbauer, 2018-03-21:
@winash12 OK, I think we talk different things. Do you use 2D near surface products or 3D products? The matching stuff @vlouf and me were referring to is in 3D.

If you just want to use 2D near surface products than you are almost ready. Just getting the swath 2d data and use the provided lat lon coordinates. But I'm possibly missing something.

winash12, 2018-03-21:
@kmuehlbauer No I need th 3D Product I am interested in studying mesoscale convective systems . So I need the 3D echo.

winash12, 2018-03-21:
@kmuehlbauer So both of NCAR's SPRINT and REORDER(no longer supported)  I believe use bilinear interpolation from radar coordinate system to the xyz Cartesian grid.

kmuehlbauer, 2018-03-21:
@winash12 What is your destination resolution of your xyz grid? The SR 'grid' is quite huge. Also, for GR comparison one would assume GR centric aeqd. Over ocean and the like you would need to fix your xyz grid at some point.

winash12, 2018-03-21:
@kmuehlbauer I think that would resolution would be determined by data storage computations as you rightly point out that the SR 'grid' is huge. So 2 km by 2 km in the horizontal and 0.5 km in the vertical. The total interpolation grid would  be around 100 km x 100 km x 12 kms. 

@scollis I will definitely ping @dopplerchase offline