ARM-DOE/pyart #720: texture_of_complex_phase doesn't like non-ARM variable names

swnesbitt, 2018-03-25:
```
phi_txt=pyart.retrieve.texture_of_complex_phase(r1, phidp_field='PHIDP')
```
on non-ARM data with a different variable name for measured differential phase gives
```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-5-e5c76bcaaab5> in <module>()
      5 gatefilter.exclude_above('TRIP_FLA', 2)
      6 vel_txt=pyart.retrieve.calculate_velocity_texture(r1, vel_field='VEL')
----> 7 phi_txt=pyart.retrieve.texture_of_complex_phase(r1, phidp_field='PHIDP')
      8 
      9 fig = plt.figure(figsize=(15, 9))

~/anaconda/lib/python3.6/site-packages/pyart/bridge/wradlib_bridge.py in texture_of_complex_phase(radar, phidp_field, phidp_texture_field)
     69 
     70     # Grab the phase data
---> 71     phidp = radar.fields[phidp_field]['data']
     72 
     73     # convert to complex number

KeyError: 'differential_phase'
```

scollis, 2018-03-25:
Thanks @swnesbitt @rjackson will lake a look

scollis, 2018-03-25:
Oops I meant Bobby

rcjackson, 2018-03-26:
This is due to a bug in the code when phidp_texture_field is None (the default setting). It just sets the phidp_field to the ARM name when you set phidp_texture_field to None. I put in a PR fixing this and adding better metadata.

scollis, 2018-04-03:
Closed by #721

closed 2018-04-03 by scollis