ARM-DOE/pyart #726: issue with _interpolate_azimuth_edges

meteoswiss-mdr, 2018-04-10:
Hi,

I encountered a very weird issue with the function _half_angle_complex in transforms.py.

If the azimuth angles of the rays are all the same, due to rounding the dot_product may be larger than one and therefore produce a nan value. In the newest versions of matplotlib pcolormesh cannot handle masked values for the x or y arguments and therefore it raises an exception.

I solved the issue simply imposing that the dot_product cannot be larger than one:

def _half_angle_complex(complex_angle1, complex_angle2):
    """
    Return half the angle between complex numbers on the unit circle.

    Parameters
    ----------
    complex_angle1, complex_angle2 : complex
        Complex numbers representing unit vectors on the unit circle

    Returns
    -------
    half_angle : float
        Half the angle between the unit vectors in degrees.

    """
    dot_product = np.real(complex_angle1 * np.conj(complex_angle2))
    if dot_product > 1:
        warnings.warn('dot_product larger than one')
        dot_product = 1.
    full_angle_rad = np.arccos(dot_product)
    half_angle_rad = full_angle_rad / 2.
    half_angle_deg = np.rad2deg(half_angle_rad)
    return half_angle_deg

You can test the issue by checking what happen with angle 274. with regular float precision.

I hope I am clear.

scollis, 2018-04-10:
Thanks @meteoswiss-mdr ! @zssherman can you handle this.

zssherman, 2018-04-10:
@scollis Sounds good, I will take a look. Thanks @meteoswiss-mdr for catching this!

zssherman, 2018-04-10:
@scollis After a short discussion with @rcjackson and after I ran the unit tests, seems good. @meteoswiss-mdr I can make the changes or if you want to do a pull request? Either or.

meteoswiss-mdr, 2018-04-11:
It is a minor change. For me there is no need for a pull request. You can make the changes yourself. Thank you!

zssherman, 2018-04-11:
Sounds good to me, I will do that.

scollis, 2018-04-11:
Fixed by #729

closed 2018-04-11 by scollis